arduino-libraries/Ethernet #1: Library Manager update Bug

Testato, 2015-12-29:
today i found a several bug on the update system

- installed the IDE 1.6.7 in "portable" version
- Library Manager - Type Updatable
- appear: Ethernet - SD - Servo libraries to update

if you try to update the version become the "Second Last" not the "Last"

If you try to re-update the version become your "Old" version, not the "Last"

There is no way to update to the last version on this three libraries (i do not know if this bug affect other libraries also)

per1234, 2015-12-29:
Which board do you have selected in Tools > Board when you encountered this issue? This problem is caused by the libraries bundled with the board package having the same name in library.properties as the Built-In Arduino library. This confuses Library Manager. The solution is to change the bundled library name in library.properties to a unique value.

Testato, 2015-12-29:
you are right,
i have the Adafruit Huzzah board selected, whit any Arduino board the problem is not present.
So who must correct this ? it's an Arduino problem ? Adafruit problem ? Third part Core problem?

per1234, 2015-12-29:
I consider it to be a bug with the board package, I'm guessing you're using the esp8266 package so in this case I think it's their responsibility to fix it. When you think about it these libraries should not have the same name as the Arduino built in libraries because the modifications make them different libraries. Of course the header file name must be the same so that they are a drop in replacement but the name in library.properties should be changed. I submitted a pull request to fix this issue a month ago to the esp8266 repository but it hasn't been merged: https://github.com/esp8266/Arduino/pull/1096.

Testato, 2015-12-29:
I agree whit you, this is an important step on all third part core
thanks for you answer

PaulStoffregen, 2015-12-29:
The library manger has a terrible bug for 3rd party boards and libraries.  It always installs to the sketchbook libraries folder.  When libraries are installed to that location, they always override all others.

The problem happens when a 3rd party board provides its own library, like Ethernet.  The Arduino IDE allows 3rd party boards to override the normal libraries, which allows 3rd party board like ESP8266 to have their own Ethernet library with necessary modifications.

When the libraries manager installs a non-ESP8266 version of Ethernet, that copy overrides the normal Ethernet library and also ESP8266's customized library.

per1234, 2015-12-29:
@PaulStoffregen yes I agree that is a very serious [issue](https://github.com/arduino/Arduino/issues/4064) that affects many boards packages. I have [notified](https://github.com/esp8266/Arduino/issues/1097) esp8266 about the override issue so they are aware of it as a likely cause of future issue reports. It's very frustrating that the Arduino developers don't seem to have any intention to fix this problem although it even affects Arduino certified boards like Galileo, Edison, and 101. The issue reported here is not caused by that problem but can exacerbate it because it causes users to receive updatable library notifications when they actually have the most recent version already installed in the Arduino IDE install folder. If they "update" to the previous version then it is installed to the sketchbook where it causes the override issue.

PaulStoffregen, 2015-12-29:
This isn't a simple thing to fix.

mbanzi, 2015-12-30:
@per1234 why do you say that "Arduino developers don't want to fix this"? Where did you get this impression?

per1234, 2015-12-30:
@mbanzi well you're misquoting me but that's the impression I get from reading the issue report(https://github.com/arduino/Arduino/issues/4064) which was reported two months, and two IDE versions, ago. The issue has been in existence since whenever Library Manager updates of Built-in  libraries began. I'd be happy to discuss it more, and especially would appreciate any opinions on the solution I proposed, but I'd prefer to do so in the relevant issue report as it's off topic here. This issue has been resolved and should be closed.

PaulStoffregen, 2015-12-30:
I believe it's important to keep some perspective here.  Two months can seem like a long time, but it really isn't in the grand scheme of things.  Throughout Arduino's long history, many issues have persisted for much longer.  Some a **lot** longer....

Arduino clearly does want to support 3rd party boards.  They've gone to a tremendous amount of work to build a very flexible system to enable this.  But speaking as the maker of one of the very widely used 3rd party boards, I believe it can also be said issues that mostly affect 3rd party boards do get a lower priority.  I would say Arduino probably does want to fix this and other issues, but given limited resources and only so many hours in every day, it probably ends up a lower priority.

On this specific issue, I've been working on a patch that prints an additional message when opening examples from libraries which have conflicts.  Currently those messages are printed during compilation, but they can be easily missed or mistaken for regular compiler output.  Another message won't actually fix this problem (where the library manager creates conflicts), but it can at least help people to understand the conflict exists.  It's also something doable in a short time frame.  Changing the library manager is a much larger, much harder task that's going to take someone *far* more work.  I might work on this eventually, but like everyone else, I have to prioritize my development efforts.....

soundanalogous, 2016-01-16:
In addition to printing a console message, could a message be displayed in the library manager when a user attempts to update a core library? Something along the lines of:

"Warning, updating [library name] may break compatibility with third party boards. If you have issues when compiling for a 3rd party board that uses [library name], delete or temporarily remove [library name] from your sketchbook libraries directory and recompile". 

Since that is what is required when conflicts are encountered, especially for people who use a mix of core Arduino boards and third-party boards.

closed 2019-05-12 by per1234