arduino-libraries/Ethernet #4: EthernetServer begin returns void

agdl, 2016-07-08:
_From @vprheli on June 8, 2016 11:22_

Hi, Why the function EthernetServer.begin(port|null) returns void? It calls function socket() which return uint8_t as result, same as EtherentUdp.If sock counter > MAX_SOCK_NUM it fails also.Or add some function like isListening() to know if it socket is in listening state.

_Copied from original issue: arduino/Arduino#5023_

PaulStoffregen, 2018-07-23:
This is meant to solve the need to know if the server really is listening for new clients.

https://github.com/PaulStoffregen/Ethernet/commit/5d5588a160b4a18a0895678da0f2d37de7fa5bcb

Changing the return type of begin() requires editing Server.h in every core library.  Not easy....

PaulStoffregen, 2018-07-27:
Fixed by bool operator

closed 2018-07-27 by PaulStoffregen