arduino-libraries/Ethernet #10: Ethernet - Source port value bug

agdl, 2016-07-12:
_From @jonface on November 7, 2014 23:31_

Arduino v1.5.8

Problem: The source port for a client TCP connection is fixed to start from 1024. If the micro controller is connect to a NAT gateway, the gateway will store this value. When the micro controller is power cycled (so no connection tear down) and tries to make a connection, once again the source port will be 1024. The NAT gateway will be confused (by the previous connection in it's connection table) and the connection will not start until the source port is incremented. 

Fix: In EthernetClient.cpp, function,

` int EthernetClient::connect(IPAddress ip, uint16_t port)`

Changed the source port so it is set randomly,

`randomSeed(analogRead(0));`
`_srcport = 1024 + random(1, 60000);`

I'm sure there are better ways than this, but this connects 'instantly' now however many times I power cycle the controller.

_Copied from original issue: arduino/Arduino#2425_

agdl, 2016-07-12:
_From @matthijskooijman on November 8, 2014 14:42_

Hm, interesting. I wonder if there are any official recommendations (e.g. RFCs) on picking a source port...

agdl, 2016-07-12:
_From @jonface on November 8, 2014 14:57_

Closest I've found, https://tools.ietf.org/html/rfc6056 , I can't find anything which mentions NAT gateways in particular, maybe we are just meant to know this.

The fix does seem to work well, the randomness is random enough, the upper port range needs adjusting though, I was feeling lazy.

stefannilsson, 2017-03-23:
Any updates on this issue?

PaulStoffregen, 2018-07-23:
I added a patch for this on my Ethernet fork, which is soon to become version 2.0.

https://github.com/PaulStoffregen/Ethernet/commit/9454f3c9923cee2150150d25f0e26beb07cd7e7d

However, it only works when using DHCP, where the timing of the response gives some entropy.  Still no solution for static settings.  Not sure if analogRead is a good idea?

PaulStoffregen, 2018-07-27:
This is now fixed when using Ethernet.begin(macaddr) for DHCP config.

Still no resolution for non-DHCP - how to reliably get entropy??