arduino-libraries/Ethernet #14: A proposal for more than one client in the Ethernet Shield acting as a server.

agdl, 2016-07-12:
_From @carlosdelfino on February 1, 2013 16:13_

Hello

Enjoying the post of @DavyLandman, I would make a small change I made in the function:

EthernetClient EthernetServer :: available ()

EthernetServer class.

The change function gives the possibility of obtaining more than one customer when using a server. Every call this having a new customer is returned, reserving for the sock that is not used until it is released when the customer receives the signal stop (), this releases the sock is worn allowing its use again.

I am doing some tests with my fork Ethernet library and the Arduino which focuses on the use of threads based on DuinOS / FreeRTOS, and would like to invite you to test my changes together with those already made​​, so I can also their experience.

I realized that the current library does not allow more than one client concurrently, and then I made a small change that is not yet finished and well tested, but it is helping me in my tests.

The code is not yet final, but is enough for a proposal for a new approach in obtaining the clients connected to a server.

Principal function affected:
    EthernetClient EthernetServer::available()
    {
      accept();
    
      for (int sock = 0; sock < MAX_SOCK_NUM; sock++) {
    	  if(Ethernet.isSocketFree(sock)){
    		  EthernetClient client(sock);
    		  if (EthernetClass::_server_port[sock] == _port &&
    				  (client.status() == SnSR::ESTABLISHED ||
    				  client.status() == SnSR::CLOSE_WAIT)) {
    			  if (client.available()) {
    				  // XXX: don't always pick the lowest numbered socket.
    				  Ethernet.useSocket(sock);
    				  return client;
    			  }
    		  }
    	  }
      }
    
      return EthernetClient(MAX_SOCK_NUM);
    }

File Changed:
https://github.com/carlosdelfino/Arduino/blob/ide-1.5.x/libraries/Ethernet/Ethernet.cpp 
https://github.com/carlosdelfino/Arduino/blob/ide-1.5.x/libraries/Ethernet/Ethernet.h
https://github.com/carlosdelfino/Arduino/blob/ide-1.5.x/libraries/Ethernet/EthernetClient.cpp
https://github.com/carlosdelfino/Arduino/blob/ide-1.5.x/libraries/Ethernet/EthernetServer.cpp
https://github.com/carlosdelfino/Arduino/blob/ide-1.5.x/libraries/Ethernet/EthernetServer.h

ps.
@DavyLandman, I removed the previous post and created a new discussion.

_Copied from original issue: arduino/Arduino#1260_

agdl, 2016-07-12:
_From @allenhuffman on March 8, 2014 5:10_

I ran in to the issue of not supporting multiple server connections properly when I first got an Ethernet shield in 2013. I hacked my local copy to resolve the issue, and can now handle four incoming connections. I would love to compare notes. This bug is something that really should be fixed. (http://subethasoftware.com/2013/04/09/arduino-ethernet-and-multiple-socket-server-connections/)

q2dg, 2016-08-04:
Well...come on...

q2dg, 2016-08-04:
Another and (much easier) workaround is this, which I've written based on Advanced Chat Server example. For me it's fine:

`#include <Ethernet.h>
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; 
byte ip[] = { 192, 168, 1, 177 };                    
EthernetServer miservidor(23);
EthernetClient misclientes[4];         
void setup() {
  Ethernet.begin(mac, ip);                                     
  miservidor.begin();                                          
}
void loop() {
  EthernetClient uncliente = miservidor.available();   
  if (uncliente.connected() == true) { 
    boolean esClienteNuevo=true;
    for (byte i= 0 ; i < 4; i++) {
      if(uncliente == misclientes[i]) {
        esClienteNuevo = false;
        break;
      }
    }
    if (esClienteNuevo == true){
      for (byte i=0; i<4; i++){
           if (misclientes[i].connected() == false) {
             misclientes[i]=uncliente;
             uncliente.print("Hola, cliente número:");
             uncliente.println(i);
             break;
        }
      }
    }
    for(byte i=0; i<4; i++){
      while (misclientes[i].available()>0){
         char dato=misclientes[i].read();
         misclientes[i].write(dato);
      }
      if (misclientes[i].connected() == false) { 
        misclientes[i].stop();
      }
    }
  }
}`