arduino-libraries/Ethernet #19: Add printable for MAC addresses in Ethernet library [imported]

agdl, 2016-07-12:
_From @cmaglie on November 15, 2012 18:52_

This is [Issue 840](http://code.google.com/p/arduino/issues/detail?id=840) moved from a Google Code project.
Added by 2012-03-02T22:45:01.000Z by [tom.i...@gmail.com](http://code.google.com/u/102867599453860860098/).
Please review that bug for more context and additional comments, but update this bug.

Original labels: Type-Enhancement, Priority-Medium

### Original description

**What change would like to see?**
 It'd be useful to be able to print MAC addresses as well as IP addresses.

_Copied from original issue: arduino/Arduino#840_

agdl, 2016-07-12:
_From @q2dg on August 21, 2015 10:11_

As it does https://www.arduino.cc/en/Reference/WiFiMACAddress...

per1234, 2022-02-22:
I think it is off topic for this repository. The `Printable` should be implemented in the `arduino/ArduinoCore-API` repository, just as `IPAddress` class is now:

https://github.com/arduino/ArduinoCore-API/blob/master/api/IPAddress.h

closed 2022-02-22 by per1234