arduino-libraries/Ethernet #24: DHCP client doesn't handle multiple DNS server addresses in DHCP response [imported]

agdl, 2016-07-12:
_From @cmaglie on November 15, 2012 18:39_

This is [Issue 569](http://code.google.com/p/arduino/issues/detail?id=569) moved from a Google Code project.
Added by 2011-07-07T20:57:41.000Z by [adrian.m...@gmail.com](http://code.google.com/u/108388509099941515037/).
Please review that bug for more context and additional comments, but update this bug.

Original labels: Type-Defect, Priority-Medium, Component-Core

### Original description

With the new Arduino 1.0 Ethernet library with the DHCP client included, if your DHCP server returns more than one DNS server address for use it fails to parse the remainder of the DHCP response correctly.

This often results in the Ethernet chip not being configured correctly (which can sometimes be spotted if the netmask and/or gateway IP address are set to 0.0.0.0) and the Arduino won't be able to communicate over Ethernet

_Copied from original issue: arduino/Arduino#569_

agdl, 2016-07-12:
_From @powtac on April 6, 2013 14:45_

Very interesting! I research this problem since days. On Wireshark I see that the DHCP ACK goes back to 0.0.0.0 And I could not explain why. How can I check if my DHCP server returns more than one IP?

Any solutions @cmaglie ?

woytam, 2022-02-22:
I think, this issue is no longer actual as second and later DNS in DHCP response is discarded on line
https://github.com/arduino-libraries/Ethernet/blob/993eea87da7f6716b0c2b2be848476670a2fab8c/src/Dhcp.cpp#L295