arduino-libraries/Ethernet #25: Ethernet shield should have pullup resistors on CS signals

agdl, 2016-07-12:
_From @PaulStoffregen on December 1, 2014 1:19_

The Arduino Ethernet shield should be revised, to add pullup resistors on the 2 chip select signals.

Without resistors to keep the CS signals high (inactive), the SD card can "hear" communication when only the Ethernet library is used.  The card can even output data on the MISO line while the W5100 ethernet chip running, causing corruption.  See this issue for confirmation this problem does indeed happen in the wild.

https://github.com/arduino/Arduino/issues/2168

Shields should be designed with pullup resistors, to make them easy to use for novice Arduino users.  These problems are very difficult for Arduino users to resolve, but very easy to prevent by adding inexpensive resistors.  Here's an article I wrote recently with more detail.

http://www.dorkbotpdx.org/blog/paul/better_spi_bus_design_in_3_steps

Please consider revising the ethernet shield to add two pullup resistors on the chip select lines.

_Copied from original issue: arduino/Arduino#2478_

agdl, 2016-07-12:
_From @q2dg on August 16, 2015 0:9_

Well, now there is a big reshuffle of all production line, it should be THE moment for that

PaulStoffregen, 2018-07-23:
The new MKR ETH shield has 10K pullups on the W5500 and SD card CS signals, so I'm closing this old issue.

closed 2018-07-23 by PaulStoffregen