arduino-libraries/Ethernet #30: much faster Print.cpp on ethernet

agdl, 2016-07-12:
_From @pidloop on April 26, 2016 20:42_

Hello. I greatly improved the speed of my W5100 ethernet shield by adding some buffering to the method in Print.cpp for FLASH memory strings. This wiki won't let me attach it here but you can get it from my web site at http://www.clearskyinstitute.com/ham/AST or contact me at ecdowney at clearskyinstitute.com. Cheers.

_Copied from original issue: arduino/Arduino#4899_

PaulStoffregen, 2018-07-23:
I believe this issue properly belongs on the core library (for each platform), since Print.cpp isn't part of the Ethernet library.

cmaglie, 2018-07-23:
Indeed it depends on the cores. Moreover it's very old and mostly solved.

closed 2018-07-23 by cmaglie