arduino-libraries/Ethernet #31: Ethernet.begin(mac) hangs

agdl, 2016-07-12:
_From @nmklotas on January 21, 2015 22:26_

Hello,

I use Arduino uno rev3 board on windows 8.1 x64 with Arduino 1.58 beta, and after uploading the sketch, code line: 
<code>
Ethernet.begin(mac);
</code>
hangs and does not return! Been waiting for like 5 - 10 mins.

Also setting pin 4 as an output and writting HIGH to it didn't help.
(I found this instruction here http://arduino.cc/en/Main/ArduinoEthernetShield)

Not working example would be in Arduino IDE > File > Examples > DhcpAddressPrinter

Thank you

_Copied from original issue: arduino/Arduino#2565_

agdl, 2016-07-12:
_From @Julius2342 on December 27, 2015 7:32_

Hi @nmklotas, how did you solve this issue?

Best

Julius

PaulStoffregen, 2018-07-27:
Going to close this old issue.  Pretty sure the old MR reset code was where things could hang up.  Now we have W5100Class::softReset() which properly gives up after trying 20 times.

closed 2018-07-27 by PaulStoffregen