arduino-libraries/Ethernet #32: Add possibility to change server tcp port in void setup. No more fixed port!

agdl, 2016-07-12:
_From @5a2v0 on June 22, 2016 14:53_

Before this mofication:

EthernetServer server(80);
void setup() {
  //some stuff here
  server.begin(); //User can't change initial port value
  //other stuff here
}

Now:

EthernetServer server; //without port declaration but still working old declaration system...
void setup() {
  //some stuff here
  unsigned int tcp = EEPROM.read(x)
  server.begin(tcp); //User can now use a different port, maybe saved in eprom directly from the sketch in a configuration page that permise to choice a value for the tcp port...
  //other stuff here
}

Many thanks to Arduino's forum user: SukkoPera

_Copied from original issue: arduino/Arduino/pull/5062_

agdl, 2016-07-12:
_From @sandeepmistry on July 12, 2016 13:0_

Hi @5a2v0,

Thank you for submitting this change. Could you please resubmit a pull request to this repo:  https://github.com/arduino-libraries/Ethernet

Then close this pull request. The `libraries/Ethernet` folder is synced with the other repo. I have linked to.

closed 2016-07-12 by agdl

q2dg, 2016-08-02:
Maybe this issue should be reopen here again