arduino-libraries/Ethernet #6: Ethernet Shield presence detection

agdl, 2016-07-12:
_From @ibantxo on January 7, 2016 21:6_

Hi,

I want that my code works in 3 cases:
1) Without ethernet shield connected to Arduino.
2) With ethernet shield, but unpluged cable
3) With ethernet shield with a working network

The third scenario works well. But the first one "hangs" in this code line (waiting forever...):
ethClient.connect(server, port);

Is there any way to detect presence of ethernet shield or to insert a timeout in connection attempt?

Many thanks

Iban

_Copied from original issue: arduino/Arduino#4400_

agdl, 2016-07-12:
_From @mikaelpatel on January 7, 2016 22:37_

The W5100 Ethernet controller has a very specific SPI protocol which actually makes it easy to detect if a shield is present. Please see the product data-sheet for more details. Cheers! Mikael

agdl, 2016-07-12:
_From @PaulStoffregen on January 7, 2016 23:36_

You should check the return value from Ethernet.begin().  If it returns false, you should never call any of the other functions!  Of course you can expect them to hang or have unpredictable results if you got a false from Ethernet.begin().

agdl, 2016-07-12:
_From @ibantxo on January 8, 2016 14:14_

Umm...
Thanks Paul. But Ethernet.begin returns "void" when Ethernet.begin(mac,ip) is used.
Anyway... I have been using "Ethernet.begin(mac,ip)" inside void setup(void). Could it be used inside main loop or inside a function? Or is that a "initial setting"?

If "Ethernet.begin" could be outside "setup", another way I am thinking about is a "reconnect" function (called from main loop):

boolean reconnect(void) {
  Ethernet.begin(mac, ip);

  Serial.println("RECONNECT!");
  int status=ethClient.connect(server, port);
  Serial.print("Status: ");
  Serial.println(status, DEC);
  ethClient.stop();

and test with "status" value to return...

Thanks mikaelpatel. I am newbie. I will try to test "SPI way" as well. But I think is a "deeper code" for my skill level.

agdl, 2016-07-12:
_From @mikaelpatel on January 8, 2016 20:44_

@ibantxo 
My comment was mainly for the library maintainers as the board detect could be improved without forcing usage of DHCP. If you would like to do a detect to configure your sketch this could be done by a simple function that reads a register (byte) from the controller. The SPI sequence is very unique and this will help determine if a device is available before initiating etc.

agdl, 2016-07-12:
_From @ibantxo on January 8, 2016 22:32_

That could be great, Mikael. Knowing the "status" of the device:

- if the device is present/available or not
- ethernet link status is ok, or not
....

Many thanks!!!

PaulStoffregen, 2018-07-23:
Version 2.0.0 will have Ethernet.linkStatus() and Ethernet.hardwareStatus().

PaulStoffregen, 2018-07-27:
Fixed

closed 2018-07-27 by PaulStoffregen