arduino-libraries/Ethernet #9: Socket implementation in Ethernet library is prone to buffer overflow.

agdl, 2016-07-12:
_From @josephlm on March 11, 2015 22:52_

The code in recvfrom() in libraries/Ethernet/src/utility/socket.cpp does not check the length of the buffer (buf) passed before copying data from Wiznet shield buffer into the application's passed buffer.

If 'data_len' is bigger than 'len' then the only 'len' bytes should be copied to the application's passed buffer, otherwise 'data_len' bytes should be copied to the application's passed buffer.

_Copied from original issue: arduino/Arduino#2756_

PaulStoffregen, 2018-07-23:
Version 2.0 will remove recvfrom().  EthernetUDP to use safe socketRecv().

(this issue to be closed with 2.0 published)

PaulStoffregen, 2018-07-27:
Fixed

closed 2018-07-27 by PaulStoffregen