arduino-libraries/Ethernet #41: Authors and License issues

cmaglie, 2016-08-30:
The Ethernet library has a big amount of external contributions, the list of all contributors should be put in a CONTRIBUTIONS file and the library.properties adjusted accordingly.

Moreover there are a bunch of files without a license, these files should be fixed by adding a proper license header.

See also https://github.com/arduino/Arduino/issues/5272#issuecomment-243426974

MisterAwesome23, 2020-03-22:
I would like to work on this. Is it still open?

per1234, 2020-03-23:
This part:
>the list of all contributors should be put in a CONTRIBUTIONS file and the library.properties adjusted accordingly.

was fixed by https://github.com/arduino-libraries/Ethernet/commit/7e8debc45fab5e9b03f1e562fb6181967334c101

I think there have been some new contributors since then, but IMO if they wanted to be added to the list they should have done that as part of their PR.

Regarding this part:
>there are a bunch of files without a license, these files should be fixed by adding a proper license header.

I'm not at all an expert on this subject, but it looks like the only source files without a license are:
- [src/Dhcp.cpp](https://github.com/arduino-libraries/Ethernet/blob/master/src/Dhcp.cpp)
- [src/Dhcp.h](https://github.com/arduino-libraries/Ethernet/blob/master/src/Dhcp.h)

The only examples **with** licenses are:
- [LinkStatus](https://github.com/arduino-libraries/Ethernet/blob/master/examples/LinkStatus/LinkStatus.ino)
- [UDPSendReceiveString](https://github.com/arduino-libraries/Ethernet/blob/master/examples/UDPSendReceiveString/UDPSendReceiveString.ino)
- [UdpNtpClient](https://github.com/arduino-libraries/Ethernet/blob/master/examples/UdpNtpClient/UdpNtpClient.ino)
- [WebClientRepeating](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebClientRepeating/WebClientRepeating.ino)

I have no idea what the process would be for resolving the licensing issue.

MisterAwesome23, 2020-03-23:
Oh. Thanks for pointing out the specifics.
Do you want me to dive in to know more about the process of licensing?

P.S: I am a fellow Google Summer of code 2020 aspirant. Kindly let me know if i can be of any help. @per1234