arduino-libraries/Ethernet #42: client.connect() fails after a while

florentvaldelievre, 2016-10-17:
**System:** Ethernet lib 1.1.2, Ethernet shield, Ethernet shield, arduino IDE 1.6.12
**Description:** client.connect() returns 0 after several successful POST requests

It seems to be exactly the same error as [ISSUE-15](https://github.com/arduino-libraries/Ethernet/issues/15). Unfortunately, the proposed code does not solve the problem on my side.

**Detailed description:** I am sending small amount data into an elasticsearch server using POST request every 2 seconds. After a while, could be 15 minutes, could be hours, Ethernet client does not manage to connect to my server ( Server up and running for sure ). At this step, The ATMega is not pingable anymore. The only way I have found to make it working again is to unplug/plug the power source.

Ive made sure I have the following line in EthernetClient.cpp as mentioned in http://forum.freetronics.com/viewtopic.php?t=176

```
int EthernetClient::connect(IPAddress ip, uint16_t port) {
  if (_sock != MAX_SOCK_NUM)
    return -1;

  for (int i = 0; i < MAX_SOCK_NUM; i++) {
    uint8_t s = W5100.readSnSR(i);
      if (s == SnSR::CLOSED || s == SnSR::FIN_WAIT || s == SnSR::CLOSE_WAIT) {
      _sock = i;
      break;
    }
  }
```

And you will find below the code:
```

#include <SPI.h>
#include <Ethernet.h>
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
EthernetClient client;
#include <TimeLib.h>

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  Ethernet.begin(mac);
  Serial.println(Ethernet.localIP());
}

void loop() {

 String dataES = "{\"location\": \"office\"}";
 if(!postPage("192.168.1.68",9200,"/sensors/temperature",dataES)) 
  Serial.print(F("Fail "));
 else {
  Serial.print(F("Pass "));
  Serial.println(dataES);
 } 
 delay(1000);
}  

byte postPage(char* domainBuffer,int thisPort,char* page, String thisData)
{
  int inChar;
  char outBuf[64];
  int TIMEOUT_REPLY = 5000;
  Serial.print(F("connecting..."));

  if(client.connect(domainBuffer,thisPort) == 1){
    Serial.println(F("connected"));
    sprintf(outBuf,"POST %s HTTP/1.1",page);
    client.println(outBuf);
    sprintf(outBuf,"Host: %s",domainBuffer);
    client.println(outBuf);
    client.println(F("Connection: close\r\nContent-Type: application/json"));
    sprintf(outBuf,"Content-Length: %u\r\n",thisData.length());
    client.println(outBuf);
    client.print(thisData);
  } 
  else {
    Serial.println(F("failed"));
  }

  unsigned long ctime = millis() + TIMEOUT_REPLY;
  while(client.connected()) {       
    if ( millis()>ctime ) {
       Serial.println("NO RESPONSE WITHIN PRESCRIBED TIME");
      break;
    }
    while(client.available()) {
      inChar = client.read();
      Serial.write(inChar);
      delay(1);
    }
    delay(1);
  }

  Serial.println();
  Serial.println(F("disconnecting."));
  client.stop();
  return 1;
}
```

Any idea ? I have been battling for months on this problem. Thanks !

florentvaldelievre, 2016-10-17:
@stevstrong, @agdl  Did you experience the issue ?

Rotzbua, 2017-06-12:
I guess maintain() of ethernet class in loop() is missing, because you use dhcp.

stefanofalasca, 2019-01-01:
I 've too a similar problem, using static ip. Somebody found a solution?

colinner, 2021-01-13:
I had this problem. I have the Wiznet5500 chip, static IP address. I could make 3 requests, and the 4th would always fail. Oddly, if I requested from an external IP, it would still work >3 requests, but internal IPs it would not. Maybe this was coincidence.
Anyway, the problem was my fault: I had defined EthernetClient twice, and was client.stop() ing the wrong EthernetClient.