arduino-libraries/Ethernet #45: Overflow memset in Dhcp.cpp

Rotzbua, 2017-02-08:
Overflow risk in `Dhcp.cpp`.
_dhcpLocalIp has only 4 byte.
Every variable should be filled separate with zeros.
```c++
void DhcpClass::reset_DHCP_lease(){
    // zero out _dhcpSubnetMask, _dhcpGatewayIp, _dhcpLocalIp, _dhcpDhcpServerIp, _dhcpDnsServerIp
    memset(_dhcpLocalIp, 0, 20);
}
```

GunZi200, 2020-07-02:
This out of bounds access of the buffer is still in the code. The error goes away with this code below, but I haven't tested.
```c
void DhcpClass::reset_DHCP_lease()
{
	// zero out _dhcpSubnetMask, _dhcpGatewayIp, _dhcpLocalIp, _dhcpDhcpServerIp, _dhcpDnsServerIp
	memset(_dhcpSubnetMask, 0, 4);
	memset(_dhcpGatewayIp, 0, 4);
	memset(_dhcpLocalIp, 0, 4);
	memset(_dhcpDhcpServerIp, 0, 4);
	memset(_dhcpDnsServerIp, 0, 4);
}
```

technyon, 2023-01-31:
Just noticed this when running cppcheck over the code. This code is very dangerous:

- There's no guarantee that the compiler puts these arrays into sequential memory locations, so this could potentially overwrite unrelated memory locations
- The compiler is free to change the alignment. So even if the arrays are put into sequential memory locations, there could be added bytes for padding, depending on architecture. That would result in not all bytes being zeroed
- If someone changes the code unaware of that memset (e. g. adding / removing / moving around some of the arrays), a lot bad stuff happen :)

I've created a PR to fix this.