arduino-libraries/Ethernet #47: Allow TLS use in Ethernet library

q2dg, 2017-06-05:
This request was one of multiple ones in https://github.com/arduino-libraries/Ethernet/issues/37 but I've separated here to make it more visible.

I suspect that hardware design of Ethernet shield should be changed to achieve this ...but I think this is a must to be on pair with Wifi101 shield.

Thanks

Rotzbua, 2017-06-05:
@q2dg Something like this Shield?: MS1000 + W5500  http://www.e-wbm.com/wiznet_w5500.do

q2dg, 2017-06-05:
@Rotzbua YES!

bleckers, 2020-06-12:
TLS/SSL is more application layer and not really related to the Ethernet library.

If you are wanting TLS/SSL, you should use something like this client library - https://github.com/OPEnSLab-OSU/SSLClient