arduino-libraries/Ethernet #59: Arduino MEGA 2560 + W5100 no connection after update

iridiu, 2018-08-06:
I have updated the library and now I don't have Ethernet connection, not even with Standard Firmata Ethernet. Also pin 10 is set to HIGH, which I can't change

How to downgrade to previous version?

PaulStoffregen, 2018-08-06:
Use Ethernet.init(pin) before Ethernet.begin() to configure a different pin for CS.

The old version was fixed at pin 10 and could not be changed, unless you edit w5100.h.

To go back to the old version, use the library manager (Sketch > Include Library > Manage Libraries).  Search for "Ethernet", click on the library, and use the drop-down list to select the version you want to use.  Easy peasy!

PaulStoffregen, 2018-08-07:
Can you give any info about how to reproduce the problem you encountered?

iridiu, 2018-08-10:
Hi PaulStoffregen, sorry for my delay, I will describe my problem: 
I have a PLC which is based on Arduino MEGA with W5100 Ethernet Shield and don't use pin 10. About pinout and info, you can find details at this link: https://controllino.biz/wp-content/uploads/2017/05/CONTROLLINO-MEGA-Pinout.pdf

I just got back at previus version (old) which is working fine for my setup and I have not tried to modify the new version.

Thank you very much for your response

PaulStoffregen, 2018-08-11:
I've created an issue with Controllino to ask them to update their board support library.

https://github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/20

PaulStoffregen, 2018-08-13:
Controllino has updated their board support library for compatibility with Ethernet 2.0.0.

closed 2018-08-13 by PaulStoffregen