arduino-libraries/Ethernet #61: suboptimal SPI usage

d21d3q, 2018-08-23:
Original issue [here](https://github.com/adafruit/Ethernet2/issues/28).

This issue came up when I was using [Ethernet2](https://github.com/adafruit/Ethernet2) library on Nucleo board, but same problems would show up while using this Library. 
I described original issue in [Ethernet2 repo](https://github.com/adafruit/Ethernet2/issues/28) by adafruit and also on [stm32duino](https://github.com/stm32duino/Arduino_Core_STM32/issues/257). 

I've managed to solve problem by using manual transaction control (proposed in [this PR](https://github.com/adafruit/Ethernet2/pull/31))

Since Arduino is gaining popularity on other non-avr platforms, maybe it would be worth to start optimize libraries for other platforms without affecting original usage on AVR based bards?

PaulStoffregen, 2018-08-23:
Have you looked at the Ethernet code?  It is **NOT** the same as Ethernet2.  This issue is regarding the very inefficient Ethernet2 library.

PaulStoffregen, 2018-08-25:
I'm going to close this issue, because it is clearly about the code for a different library.

Ethernet (not Ethernet2) does SPI.transaction at the socket layer.  One transaction is done for each higher-level operation, not 1 per SPI use as in Ethernet 2.

Perhaps file a bug with stm32duino, asking them to improve the SPI library?

closed 2018-08-25 by PaulStoffregen