arduino-libraries/Ethernet #62: Version 2.0.0 not working for ELECFreaks Ethernet Shield

dschnabel, 2018-08-29:
**Problem:** After upgrading from Ethernet lib 1.1.2 to 2.0.0 things stopped working.

**Board:** Arduino UNO R3
**Shield:** Arduino Compatible Ethernet Shield by ELECFreaks (https://www.robotshop.com/en/arduino-compatible-ethernet-shield.html)
**Shield Specs:** https://www.robotshop.com/media/files/zip/documentation-ef02007.zip

I ran the example sketch from https://www.arduino.cc/en/Tutorial/WebClient. It says in that example:
```
Use Ethernet.h with the Arduino Ethernet Shield or Ethernet2.h with the Arduino Ethernet Shield 2 and Leonardo Ethernet.
```
Since my Shield is V1.1 I'm using `Ethernet.h`.

The error I'm getting is: `Ethernet shield was not found.  Sorry, can't run without hardware. :(` which tells me that the WIZ5100 chip isn't being recognized:
```
Ethernet.hardwareStatus() == EthernetNoHardware
```

Version 1.1.2 of the Ethernet library is working fine, so I am confident the shield is properly connected to the Arduino. Any ideas what could be going on here?

PaulStoffregen, 2018-08-29:
Try editing w5100.h.  First, to make sure you're really editing the right file, add any syntax error and make sure it shows up when you click Verify.

Then find this line:

    #define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)

Try reducing the speed.  Maybe try 4000000 or even 2000000.  Perhaps the level shifter on this shield is simply too slow?

dschnabel, 2018-08-30:
Thanks. I made sure that I'm getting a compiler error when adding a syntax error to `w5100.h`. Then I played around with different values for the SPISettings. I tried 8000000, 4000000, 2000000, and 1000000.

The behavior has changed. The output I'm getting now is this (from sketch [WebClient](https://www.arduino.cc/en/Tutorial/WebClient?action=sourceblock&num=1)) and nothing after that:
```
Initialize Ethernet with DHCP:

```
The program seems to be stuck in this line forever (note that DHCP worked in 1.1.2):
```
Ethernet.begin(mac)
```
The first line sometimes even repeats multiple times which I assume is because the board resets itself:
```
Initialize Ethernet with DHCP:
Initialize Ethernet with DHCP:
Initialize Ethernet with DHCP:

```
Is there a diagnostic sketch that I can run or would it help to add debug statements to see where things go sideways?

PaulStoffregen, 2018-08-30:
There are many commented Serial.print lines in w5100.cpp.  Maybe uncomment all those to see more info about what it's really doing.

dschnabel, 2018-08-31:
Same output for speeds 14000000, 8000000, 4000000, 2000000:
```
Initialize Ethernet with DHCP:
w5100 init
w5100.cpp: detect W5200 chip 
Wiznet soft reset
mr=0
w5100.cpp: detect W5500 chip 
Wiznet soft reset
mr=0
w5100.cpp: detect W5100 chip 
Wiznet soft reset
mr=0
no chip :-(
Failed to configure Ethernet using DHCP 
Ethernet shield was not found.  Sorry, can't run without hardware. :( 
```

ntd-94, 2020-11-27:
hi,
any solution to this problem?
thanks