arduino-libraries/Ethernet #67: Documentation updates related to the 2.0.0 release

per1234, 2018-09-02:
# Changes to existing pages
I have also included improvement suggestions not related to the 2.0.0 release for the affected reference pages because it seems like it will be easier for the person updating the page to make all necessary changes to a page at the same time, rather than addressing multiple issue reports.

---
https://www.arduino.cc/en/Reference/Libraries
Change:
>- Ethernet / Ethernet 2 - for connecting to the internet using the Arduino Ethernet Shield, Arduino Ethernet Shield 2 and Arduino Leonardo ETH

To:
>- **Ethernet** - for connecting to the internet using the Arduino Ethernet Shield, Arduino Ethernet Shield 2 and Arduino Leonardo ETH

---
https://www.arduino.cc/en/Reference/Ethernet
Change:
>Ethernet / Ethernet 2 library
>
>These libraries are designed to work with the Arduino Ethernet Shield  (Ethernet.h) or the Arduino Ethernet Shield 2 and Leonardo Ethernet  (Ethernet2.h). The libraries are allow an Arduino board to connect to  the internet.  The board can serve as either a server accepting incoming  connections or a client making outgoing ones.  The libraries support up  to four concurrent connection (incoming or outgoing or a combination).  Ethernet library (Ethernet.h) manages the W5100 chip, while Ethernet2  library (Ethernet2.h) manages the W5500 chip; all the functions remain  the same. Changing the library used allows to port the same code from  Arduino Ethernet Shield to Arduino Ethernet 2 Shield or Arduino Leonardo  Ethernet and vice versa.
>
>Arduino communicates with the shield using the SPI bus. This is on digital pins 11, 12, and 13 on the Uno and pins 50, 51, and 52 on the Mega. On both boards, pin 10 is used as SS. On the Mega, the hardware SS pin, 53, is not used to select the W5100, but it must be kept as an output or the SPI interface won't work.

To:
>Ethernet library
>
>This library is designed to work with the Arduino Ethernet Shield, Arduino Ethernet Shield 2, Leonardo Ethernet, and any other W5100/W5200/W5500-based devices.  The library allows an Arduino board to connect to the Internet.  The board can serve as either a server accepting incoming connections or a client making outgoing ones.  The library supports up to eight (W5100 and boards with <= 2 kB SRAM are limited to four) concurrent connections (incoming, outgoing, or a combination).
>
>The Arduino board communicates with the shield using the SPI bus. This is on digital pins 11, 12, and 13 on the Uno and pins 50, 51, and 52 on the Mega. On both boards, pin 10 is used as SS. On the Mega, the hardware SS pin, 53, is not used to select the Ethernet controller chip, but it must be kept as an output or the SPI interface won't work.

---
https://www.arduino.cc/en/Tutorial/WebClient
Change:
>The results of this search are viewable as HTML through your Arduino's serial window.

To:
>The results of this search are viewable as HTML **in the Serial Monitor**.

---
Change:
>The Ethernet shield allows you to connect a WizNet Ethernet controller to the Arduino or Genuino boards via the SPI bus.

To:
>The Ethernet shield allows you to connect a **WIZnet** Ethernet controller to the Arduino or Genuino boards via the SPI bus.

---
Change:
>It uses pins 10, 11, 12, and 13 for the SPI connection to the WizNet.

To:
>It uses **the ICSP header pins and pin 10 as chip select** for the SPI connection to the **Ethernet controller chip**.

---
Change:
>The shield should be connected to a network with an ethernet cable.

To:
>The shield should be connected to a network with an **Ethernet** cable.

---
Change:
>image developed using Fritzing.

To:
>**Image** developed using Fritzing.
---
Remove:
>Please note: according to your hardware setup, you need to comment / uncomment the libraries at the beginning of the sketch. Use Ethernet.h with the Arduino Ethernet Shield or Ethernet2.h with the Arduino Ethernet Shield 2 and Leonardo Ethernet.

---
https://www.arduino.cc/en/Tutorial/WebClientRepeating

Change:
[... 1201 lines cut ...]
#### Returns
the DNS server IP address for the device (IPAddress)

#### Example
```cpp
#include <SPI.h>
#include <Ethernet.h>

byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip(10, 0, 0, 177);

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Ethernet.begin(mac, ip);

  Serial.print("The DNS server IP address is: ");
  Serial.println(Ethernet.dnsServerIP());
}

void loop () {}
```

SimonePDA, 2018-09-07:
Working on all the modifications.

Should be done in the next 24 hours.

Thanks for your massive effort!

simone

SimonePDA, 2018-09-08:
Done.

The links to stream were omitted in the see also because they are potentially not stable due to the reference new structure.

closed 2018-09-08 by SimonePDA