arduino-libraries/Ethernet #75: Hang in socket.cpp while SnSR::ESTABLISHED

mamama1, 2018-11-05:
Hi

i have written a little webserver code which delivers files of any size to the web browser (client).
However sometimes, the code just "freezes" and by putting in Serial.print statements everywhere I tracked it down to socket.cpp within this loop (line 452), where it gets stuck:

```
	/* +2008.01 bj */
	while ( (W5100.readSnIR(s) & SnIR::SEND_OK) != SnIR::SEND_OK ) {
		/* m2008.01 [bj] : reduce code */
		if ( W5100.readSnSR(s) == SnSR::CLOSED ) {
			SPI.endTransaction();
			return 0;
		}
		SPI.endTransaction();
		yield();
		SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
	}
```

readSnIR(s) gives 23(dec) 17(hex) which means SOCK_ESTABLISHED according to the W5500 datasheet.
So SnIR stays at SOCK_ESTABLISHED, what means that this while loop will never exit.
Any idea why this happens?

Thanks!

Hardware: 
- W5500 module from aliexpress
- Genuine Teensy 3.5

Software:
```
#include <Arduino.h>
#include <Ethernet.h>
#include <SD.h>

#define CS_ETHERNET			23
#define CS_SDCARD			BUILTIN_SDCARD

#define DEBUG

// SD card
Sd2Card card;
SdVolume volume;
SdFile root;

// Ethernet
EthernetClient client;
byte mac[] = { 0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02 };
EthernetServer server(80);

// HTTP settings
#define HTTP_REQ_MAX_SIZE 2000
#define HTTP_URL_MAX_SIZE 2000
#define HTTP_RSP_MAX_SIZE 1000

#define HTTPDEBUG

// HTTP buffers
char httpResponseBuffer[HTTP_RSP_MAX_SIZE] = { 0 };
char httpRequestBuffer[HTTP_REQ_MAX_SIZE] = { 0 };								// maximum http request length
char httpRequestURL[HTTP_URL_MAX_SIZE] = { 0 };

// HTTP working variables
uint32_t httpRequestLen = 0;
File dataFile;
int remaining;
char hexLength[11] = { 0 };
char hexLengthMax[11] = { 0 };
uint32_t bytesWritten = 0;

[... 327 lines cut ...]
							memset(httpResponseBuffer, 0, HTTP_RSP_MAX_SIZE);
							memcpy_P(httpResponseBuffer, html_response_404, strlen(html_response_404));
							client.write(httpResponseBuffer);
						}
					}
				}
				else 
				{
#ifdef HTTPDEBUG
					Serial.println("delivering 405");
#endif // HTTPDEBUG
					// 405
					memset(httpResponseBuffer, 0, HTTP_RSP_MAX_SIZE);
					memcpy_P(httpResponseBuffer, html_response_405, strlen(html_response_405));
					client.write(httpResponseBuffer);
				}

				delay(1);
				client.stop();
			}
		}
	}

}
```

mamama1, 2018-11-06:
Anyone got an idea? My pasted code is compileablr as-is and reproduces the issue on a teensy 3.5 right away, if an sd card with a suitable file is inserted. w5500 is needed as well, of course.

thanks!

mamama1, 2018-11-08:
I think I kind of worked around this, will have to see if this is fixed in the long term.

In socket.cpp I changed this (in function uint16_t EthernetClass::socketSend(uint8_t s, const uint8_t * buf, uint16_t len)):

from: while (freesize < ret);
to: while (freesize < 2046);

This of course massively slows things down but until now the issue didn't come up again, since I changed this. I guess some kind of delay would achieve the same but I'm not sure if there is some issue with the buffer not getting emptied fast enough so I guess the check on the freesize is better than doing some random delay.

If I slow down the second loop (while ( (W5100.readSnIR(s) & SnIR::SEND_OK) != SnIR::SEND_OK ) {) or even the high level "client.write()" in my sketch, it doesn't have any effect at all so I guess something should be changed in the first loop (uint16_t EthernetClass::socketSend(uint8_t s, const uint8_t * buf, uint16_t len)) in the socket.cpp.

I'm by no means a good enough coder to figure out what's wrong exactly and how to mitigate this most efficiently, unfortunately. Until someone much smarter than me figures this out, I hope my workaround will work for me to make things stable at least, I don't need speed anyway.

SapientHetero, 2019-01-31:
Not sure why you had the problem. I've also written a webserver that uses this library and have uploaded files as large as 3GB and just downloaded a 760,189,144 byte file stored on an SD card. I am using an Adafruit Metro M4 processor card and a Wiznet W5500 Ethernet shield. I've used a SEEED W5500 Ethernet shield too without file size-related problems.

Tthe loop you originally posted will exit when the write completes or the socket is closed. I think you may not have noticed that one of the register reads is of Sn_SR (Socket n Status Register) while the other is of SN_IR (Socket n Interrupt Register). When Sn_IR bit 4 (SEND_OK) is set, it indicates that the requested write operation has completed and the loop exits and returns the number of bytes written. If while waiting for this, the socket is closed for any reason, Sn_SR will be set to 0 (SnSR::CLOSED), causing the loop to exit and return an error (0).

```
	while ( (W5100.readSnIR(s) & SnIR::SEND_OK) != SnIR::SEND_OK ) {
		/* m2008.01 [bj] : reduce code */
		if ( W5100.readSnSR(s) == SnSR::CLOSED ) {
			SPI.endTransaction();
			return 0;
		}
		SPI.endTransaction();
		yield();
		SPI.beginTransaction(SPI_ETHERNET_SETTINGS);
	}
```

I suspect your write doesn't complete for some reason. You should take another look at what readSnIR(s) is really returning, and you can use WireShark to see what's happening on the network when your system hangs. Here's what the bits in Sn_IR mean:

[image]

While looking at your code, I noticed that if I strip out all the middle part of your server, leaving just the top and bottom few lines of code, you can see that you stop the client after each read. Since your code says, "while (client.connect()" I'm guessing that wasn't what you intended. Move the client.stop() call out of that loop if you want Connection: keep-alive to actually keep your connection alive.

```
void processWebServer()
{
  client = server.available();

  if (client)
  {
#ifdef HTTPDEBUG
    Serial.println("new client");
#endif // HTTPDEBUG
    while (client.connected())
    {
      if (client.available())
      {...

        Do Stuff

        delay(1);
        client.stop();
      }
    }
  }

}
```