arduino-libraries/Ethernet #79: ¿Ethernet Shield 2 How to make it work without pin 10 (SCSn)

franciscogimeno2000, 2018-11-27:
Hello after reviewing the information that is available, I do not see anything that will run this.

Is it possible to use the Ethernet Shield by pulling the SCSn pin of the w5500 chip to negative?

According to the manufacturer's pdf the chip can work as it is now with the active signal from scsn controlled by the mega for example, but also putting it to negative and mode (Fixed Length Data Mode (SCSn is always connected by Ground)

I want to use the network module without the sd or any other device on the spi bus.
With which the selection pin would not be necessary.

Could you help me do this?

Greetings and thank you

franciscogimeno2000, 2018-11-27:
That is mine, but we do not advance.

franciscogimeno2000, 2018-11-27:
We do not see how to do it.

PaulStoffregen, 2018-11-27:
This is not supported.

closed 2018-11-27 by PaulStoffregen