arduino-libraries/Ethernet #80: Ping

MajicCat, 2018-12-07:
Has anyone got ICMP ping working?

The older examples look to have issues.

taylor-an, 2019-01-09:
[image]
[image]

It works.

MajicCat, 2019-01-09:
Is the example code posted?

taylor-an, 2019-01-10:
I'm just execute on Arduino (Windows IDE), Example > Ethernet -> ChatServer.

The hercules send a ICMP Ping.

sbminimalistics, 2019-04-28:
I know it's not an issue with the library, but I feel here I might find some ninjas to answer this question:
what about pinging the opposite direction (from Arduino to network)? ICMP Ping library doesn't work with Ethernet 2.0 any more. Tried adjusting ICMP Ping library myself, but the code looks to be over top of my understanding.
Anyone around who has already solved this puzzle?

thekunalsaini, 2020-03-30:
Hi @MajicCat ,
I'm executed it on Arduino (Windows IDE) and also uses ChatServer.

The Hercules sends an ICMP Ping.

orichienal, 2021-01-29:
I'm definitely looking for a way to ping from the arduino with ethernet 2.0, has anyone found a solution?