arduino-libraries/Ethernet #81: REALLY NEED HELP: CAN'T SEND DATA FROM NODEMCU TO DATABASE

edewarde1, 2018-12-07:
hello guys, I’m working on a school project on iot, which is based on RFID RC522 and nodemcu (esp8266), but I have a problem, semms that the data getting from nodemcu and that must be read and sent by php code not getting to data base in localhost  I do not know how to solve this, if you can help me I would be very grateful
***the system:
ESP8266 NODEMCU V1.0 + RFID RC522

WAMP SERVER V3.1.4

PHP 7.2.10

APACHE 2.4.35

MYSQL 5.7.23

***the system's arduino code:
`#include<SoftwareSerial.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <SPI.h>
#include <MFRC522.h>

const char* ssid = "ff";// 
const char* password = "ff";
//WiFiClient client;
char server[] = "192.168.10.55";   //eg: 192.168.0.222
#define SS_PIN D4 //FOR RFID SS PIN BECASUSE WE ARE USING BOTH ETHERNET SHIELD AND RS-522
#define RST_PIN D8
#define No_Of_Card 3
#define LED_PIN D2
#define LED_CIN D1
 WiFiClient client;

//WiFiServer server(80);
SoftwareSerial mySerial(8,9);     
MFRC522 rfid(SS_PIN,RST_PIN);
MFRC522::MIFARE_Key key; 
byte id[No_Of_Card][4]={
  {224,43,39,27},             //RFID NO-1          
  {188,32,7,133}              //RFID NO-2
};
byte id_temp[3][3];
byte i;
int j=0;
void setup()
{
 Serial.begin(115200);
  delay(10);
  mySerial.begin(115200);
  SPI.begin();
  rfid.PCD_Init();
  pinMode(LED_PIN,OUTPUT);
  pinMode(LED_CIN,OUTPUT);
    for(byte i=0;i<6;i++)
  {
    key.keyByte[i]=0xFF;
  }

  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
 
  // Start the server
//  server.begin();
  Serial.println("Server started");
  Serial.print(WiFi.localIP());
  delay(1000);
  Serial.println("connecting...");
 }
void loop()
{  // Check if a client has connected
  int m=0;
  if(!rfid.PICC_IsNewCardPresent())
  return;
  if(!rfid.PICC_ReadCardSerial())
  return;
  for(i=0;i<4;i++)
  {
   id_temp[0][i]=rfid.uid.uidByte[i]; 
             delay(50);
  }
  
   for(i=0;i<No_Of_Card;i++)
  {
          if(id[i][0]==id_temp[0][0])
          {
            if(id[i][1]==id_temp[0][1])
            {
              if(id[i][2]==id_temp[0][2])
              {
                if(id[i][3]==id_temp[0][3])
                {
                  Serial.print("your card no :");
                  for(int s=0;s<4;s++)
                  {
                    Serial.print(rfid.uid.uidByte[s]);
                    Serial.print(" ");
                   
                  }
                  
                  Serial.println("\nVALID");
                   digitalWrite(LED_PIN,HIGH);
                   delay(400);
                   digitalWrite(LED_PIN,LOW);
                  Sending_To_DB();
                  j=0;
                            
                            rfid.PICC_HaltA(); rfid.PCD_StopCrypto1();   return; 
                }
              }
            }
          }
   else
   {j++;
    if(j==No_Of_Card)
    {
      Serial.println("inVALID");
      digitalWrite(LED_CIN,HIGH);
      delay(400);
      digitalWrite(LED_CIN,LOW);
      Sending_To_DB();
      j=0;
    }
   }
  }
  
     // Halt PICC
  rfid.PICC_HaltA();

  // Stop encryption on PCD
  rfid.PCD_StopCrypto1();
 }

 void Sending_To_DB()   //CONNECTING WITH MYSQL
 {
   if (client.connect(server, 80)) {
    Serial.println("connected");
    // Make a HTTP request:
    Serial.println("GET /rfid/rfid_read.php?allow=");     //YOUR URL /rfid/rfid_read.php?allow
    client.print("GET /rfid/nodemcu_rfid/rfid_read.php?allow=");     //YOUR URL /rfid/rfid_read.php?allow  /var/www/html/rfid/rfid_read.php
    if(j!=No_Of_Card)
    {
      Serial.println('1');
      client.print('1');
    }
    else
    {
      Serial.println('0');
      client.print('0');
    }
    Serial.println("&id=");
    client.print("&id=");
    for(int s=0;s<4;s++)
                  {
                    Serial.println(rfid.uid.uidByte[s]);
                    client.print(rfid.uid.uidByte[s]);
                                  
                  }
    client.print(" ");      //SPACE BEFORE HTTP/1.1
    client.print("HTTP/1.1");
    client.print("Host: ");
     client.println(server);
    client.println("Host: 192.168.10.55");//eg: 192.168.0.222
    client.println("Connection: close");
    client.println();
  } else {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }
  
  client.stop();
 }`

***the system's php code:
`<?php
class rfid{
 public $link='';
 function __construct($allow, $id){
  $this->connect();
  $this->storeInDB($allow, $id);
 }
 
 function connect(){
  $this->link = mysqli_connect('localhost','*******','*******') or die('Cannot connect to the DB');
  mysqli_select_db($this->link,'pfe') or die('Cannot select the DB');
 }
 
 function storeInDB($allow, $id){
  $query = "insert into gol set rfid='".$id."', allow='".$allow."'";
  $result = mysqli_query($this->link,$query) or die('Errant query:  '.$query);
 }
 
}
if($_GET['allow'] != '' and  $_GET['id'] != ''){
 $rfid=new rfid($_GET['allow'],$_GET['id']);
}

?>
`

i hope you help guys, i'm new to embedded systems, arduino, and php , i've just found this project in internet, i have followed all the steps i'm sure that i've done everthing the same way they did it in the project but it seems that i have some configuration or something that blocks the data from getting to database, i have triyed many things, i have changed the php code, even i have initialized the variables in php code to see if any data will be sent to database but unfortunately nothing changed in the database.
ps: when  i use the server ip with the the file path and the variables(id and allow) initialized, the data goes to database but with the php code nothing changes.

help pls guys, i really need help

per1234, 2018-12-07:
First, this is not an appropriate place to ask for help with your project. This issue tracker is only to be used for reporting bugs and feature requests. The appropriate place to request help with your project is the Arduino Forum. Please read this before posting to the forum: http://forum.arduino.cc/index.php?topic=148850.

Second, you're not even using the Ethernet library!

edewarde1, 2018-12-07:
dude read please before talking, we use ethernet library when we use an ethernet cable to connect the card , i'm using WIFI -_-, i don't need it

PaulStoffregen, 2018-12-08:
No, *you* are the one who needs to read before talking!

Even if your code were using this particular library, this issue tracker is the wrong place to ask for help.  You need to ask for help on the forums, not here.

closed 2018-12-08 by PaulStoffregen

edewarde1, 2018-12-08:
what do you have instead of brain people, i'm begging for help and all you are talking about is where to post this where to ask fro that, even in the damn forum they delete my messages and says read the rules first i'm going to be examinated in this project i don't have time for your stupid rules

PaulStoffregen, 2018-12-08:
You are the one wasting other people's time, by asking for help on an issue tracker which is **not** a place where we answer questions.  This simply is the wrong place to ask questions.

Rotzbua, 2018-12-08:
@edewarde1 Invest more time in spelling and sentence building or go to twitter/facebook/whatever where someone can understand your millennial slang :rofl: