arduino-libraries/Ethernet #88: EthernetServer Error

esfx14, 2019-01-18:
Board: Adafruit ESP Feather with Ethernet FeatherWing
Upload Speed: 115200
Flash Frequency: 80 MHz

I need this EthernetServer thing to work for my programm, but even this small one here does not work and I am not able to find out what  I did wrong.

#include <Arduino.h>
#include <Ethernet.h>
EthernetServer server(80);
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}

sketch_jan17c:4:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'

 EthernetServer server(80);

                ^

In file included from C:\Users\Documents\Arduino\sketch_jan17c\sketch_jan17c.ino:2:0:

C:\Users\Documents\Arduino\libraries\Ethernet\src/Ethernet.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':

 class EthernetServer : public Server {

       ^

In file included from C:\Users\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.1\cores\esp32/Arduino.h:158:0,

                 from C:\Users\Documents\Arduino\sketch_jan17c\sketch_jan17c.ino:1:

C:\Users\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.1\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)

     virtual void begin(uint16_t port=0) =0;

                  ^

taylor-an, 2019-01-18:
Hello esfx14,

How about execute any example on arduino IDE ?

[image]

esfx14, 2019-01-18:
I've tried a few. Same problem everywhere where EthernetServer exists.

PaulStoffregen, 2019-01-18:
Try selecting "Arduino/Genuino Uno" in Tools > Boards.  Do you get the same error if you click Verify for Uno?

esfx14, 2019-01-18:
It's okay for this one. Maybe this part does not work for my kind of board ?

esfx14, 2019-01-18:
First I used version 1.0.1 and now tried 1.0.0. Neither of them works. More versions are not available.

esfx14, 2019-01-18:
Thanks. It works now.

closed 2019-01-18 by esfx14

per1234, 2019-01-18:
I'm glad to hear it's working for you. Certainly my workaround is not to be considered a resolution of the issue.

The question is how can this issue truly be resolved? Since Server.h is part of the standard Arduino API, it seems that it should be fixed in the ESP32 core. Hopefully, they will eventually start using [ArduinoCore-api](https://github.com/arduino/ArduinoCore-API/blob/master/api/Server.h). If so, they will need to work with the standard signature for this function then anyway.

per1234, 2019-01-18:
Maybe I don't understand you. I apologize in advance if the following completely misses the mark in addressing what you said:

The [ArduinoCore-api repository](https://github.com/arduino/ArduinoCore-API/blob/master/api/Server.h) contains the result of [Project Chainsaw](https://groups.google.com/a/arduino.cc/d/msg/developers/oyt_c5wePg8/_CWGyAG6CQAJ). This separates all the non-architecture specific code out of the core libraries so that the same code can be shared by all cores. This makes it much easier to maintain that code. The hope is that ArduinoCore-api will be used by the 3rd party hardware package authors in addition to its use by the official Arduino hardware packages. ArduinoCore-api is already in use in the [Arduino megaAVR Boards](https://github.com/arduino/ArduinoCore-megaavr) package.

Server.h is part of ArduinoCore-api. If Arduino core for the ESP32 eventually switches to using ArduinoCore-api, they will be using the official Server.h, rather than their non-standard Server.h.

PaulStoffregen, 2019-01-18:
Ok, I'll admit, this is my cynical mode...

Look, if every core library (Arduino's and all 3rd party including ESP32's until recently) currently has "virtual void begin() =0", and a project like ESP32 decides they want to change to "virtual void begin(uint16_t port=0) =0", what good is yet another copy of the exact same code in yet another github repository going to do?

per1234, 2019-01-18:
If they're using the code from ArduinoCore-api, the only way they can change that code is to submit a PR to the ArduinoCore-api, where it will be reviewed well enough that a breaking change like this will not get through.

They can still break compatibility of the architecture-specific parts of the core library that can't be in ArduinoCore-api, but at least this encourages standardization of some of the core API.

Even if the 3rd party hardware package authors don't see a benefit in that, I think they will still see the benefit in having a lot of their core code written and maintained for them. This is especially true for people creating new hardware packages.

Contributors benefit by being able to submit a PR to a single repository and having that change propagate across all participating hardware packages after it's merged.