arduino-libraries/Ethernet #89: Ethernet example shows data corrupted on web page and board hangs.

DelboyJay, 2019-02-02:
I have found an issue where requesting pages repetedly causes corrupted output and in some cases the board hangs forever even if no more requests are made. The corruption can be seen with the Ethernet example code.

**Hardware** 
Arduino Uno R3
W5100 Ethernet shield

**Code Version**
I am synced to this commit when testing https://github.com/arduino-libraries/Ethernet/commit/4f4ac17f296f5ffbe869730db88265ddc3e4fc15
I am using Windows 10 and Google Chrome Version 71.0.3578.98 (Official Build) (64-bit) for the browser.

**Reproduction Steps**
1. Load the Ethernet example into Arduino IDE and upload to the board.
2. Access the devices web page at [http://192.168.1.177](http://192.168.1.177)
3. Press F5 quickly and repeatedly a number of times. I have also achieved the same results by pressing F5 every second or so.
4. You will get the following corrupted output on the web page which of course is random in nature.

```
analog input 0 is 850
analog input 1 is 722
analog input 2 is 594
HTTP/1.1 200 OK is 595
analog input 4 is 1022
analog input 5 is 1022
C0Ø|"z1¿Kaþ°p\U§Æ+â°´áóH+¯&TÀPSÂJä\$à8±rÐÄõLæ`Z0;	œuT·!rUHéS;¤F/,+Ä¼Ó{.–ÒÿæðH ``ùÕ„¡	šß8)20Ç‡¾ñÒ#U&(%ò¢LÂîÈÙL€D{.¶¢[(J¯ÁKnŒ•’ß\m@òü`xAVñQ]¦óq— aôÄùüÛ0š¤‡:†qbŸ[˜Sãwü³É†£Àÿr¾o2¼<ç‰ŒÇ†áøž›Ëÿ–ÂgáåÈ®Ç€1@Hc€#­~SÊE4´¬lõËŽ—Û\,ä¶‹2ïÐ€9Ç‚Ï"IuÉ4;ŠHTTP/1.1 200 OK Content-Type: text/html Connection: close Refresh: 5 analog input 0 is 863
analog input 1 is 756
analog input 2 is 658
analog input 3 is 662
analog input 4 is 1022
analog input 5 is 1022
022
analog input 5 is 1023
analog input 0 is 893
analog input 1 is 793
analog input 2 is 686
analog input 3 is 666
analog input 4 is 1022
analog input 5 is 1022
/> analog input 2 is 599
ŠG¤.’‡Þ*tì0lØ8^Éè®DKýÇ¡žÈ»öªëß¬m_SÕfe³š~_2­3é;zÅùHTTP/1.1 200 OK î¤¹%ÛqŒ«–z' lbþÒVýlr ¡•`¸.…c¸f0h·Œ‰9âËŽIDL–ã¡-†À:Vè¸BRôE-HTTP/1.1 200 OK Content-Type: text/html Connection: close Refre841
analog input 1 is 722
analog input 2 is 634
analog iHTTP/1.1 200 OK23
(ª#ð "ûí/ô852
analog input 1 is 721
analog input 24 is 1022
analog input 1022
y­³ xrÄ6)á5TåDƒææ¾Ù–Š ¯á€€õ¤§ ø«T¬¶-×„µEn1†K	·“7±]&u—HL Ñ*ÁÈØ0‚,ï½SŠ€G“:d ­gæ0ßL6æjÀRJØyÉjî¢+š‚á1†±J/ál1æE—ØI±¦ž=þContent-Type: text/html ConnectHTTP/1.1 200 OKresh: 5 analog input 0 is 850
analog input 1 is 722
analog input 2 is 594
HTTP/1.1 200 OK is 595
analog input 4 is 1022
analog input 5 is 1022
C0Ø|"z1¿Kaþ°p\U§Æ+â°´áóH+¯&TÀPSÂJä\$à8±rÐÄõLæ`Z0;	œuT·!rUHéS;¤F/,+Ä¼Ó{.–ÒÿæðH ``ùÕ„¡	šß8)20Ç‡¾ñÒ#U&(%ò¢LÂîÈÙL€D{.¶¢[(J¯ÁKnŒ•’ß\m@òü`xAVñQ]¦óq— aôÄùüÛ0š¤‡:†qbŸ[˜Sãwü³É†£Àÿr¾o2¼<ç‰ŒÇ†áøž›Ëÿ–ÂgáåÈ®Ç€1@Hc€#­~SÊE4´¬lõËŽ—Û\,ä¶‹2ïÐ€9Ç‚Ï"IuÉ4;ŠHTTP/1.1 200 OK Content-Type: text/html Connection: close Refresh: 5 analog input 0 is 863
analog input 1 is 756
analog input 2 is 658
analog input 3 is 662
analog input 4 is 1022
analog input 5 is 1022
022
analog input 5 is 1023
...
```

greyltc, 2019-02-10:
I'm getting garbled data too, but with W5500 on an UNO. I'll try to put together a reproducible example.

DelboyJay, 2019-02-10:
Hi Grey, Can you run the standard Ethernet example for me and see if hitting F5 a number of times will cause the same problem. It would help to know if the issue still occurs for different hardware.

greyltc, 2019-02-10:
You mean [this sketch](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino)?

DelboyJay, 2019-02-10:
Yeah, that's it. I noticed that I had an SD card installed and when I removed it the corruption seems to stop. The example sketch does not use the SD card so not sure why this would happen.

greyltc, 2019-02-10:
Try explicitly deasserting the SD card's chip select line (probably pin 4) before you do anything with `Ethernet`:
```C++
#define SD_SPI_CS 4
digitalWrite(SD_SPI_CS, HIGH);
pinMode(SD_SPI_CS, OUTPUT);
```

DelboyJay, 2019-03-30:
I finally got round to checking this! This was my fault, I was using pin 4 as a watchdog strobe which was turning the SD card on and off every second! Thanks for the help.

closed 2019-03-30 by DelboyJay