arduino-libraries/Ethernet #92: W5500 loses sockets after a random time since boot, and other bugs

kwendenarmo, 2019-03-07:
Hello

I've been struggling for several months with a prototype that uses the W5500 and I can't get it to work properly. The assembly is an ATmega2560 and a W5500 and I'm working on official shields (Arduino ethernet shield 2, wiznet W5500 ethernet shield), Chinese modules and own prototypes following the datasheet and the application note. But none of them work properly, not even with the example codes and I'm going mad because I don't understand what the hell is going on. This development is an update of a previous development done with the W5100, which worked correctly (done with version 1.0.4 of the ethernet library). The final design will be running 24/7, so reliability is a requirement.

I am finding several strange bugs, which occur in an indeterminate period from the start of the program, which can be from minutes to more than a week:

- Closure of one of the open sockets, with the ability to reopen them.
- Closure of all open sockets, cannot be reopened without reset.
- Loss of communication with the W5500, requires reset.
- Random ethernet data corruption, with different incidence.
- IP badly assigned, (one is assigned and reads a totally different one).
- Always indicates the cable is not connected, when it is connected and the communication is satisfactory.

Besides, I've found other faults:

- Receive fictitious packages if compiled with arduino IDE 1.8.5, not with IDE 1.6.5.
- When opening two sockets on different ports, the library indicates that only 5 are available, and effectively does not allow more than 5 to be opened.

I am not able to isolate the source of the problem, because things that work for some boards do not work for others, there being no common nexus. As I use several mega arduino host boards, I use the same HEX program for all of them, so as not to introduce discrepancy in the different tests. I have also tried to vary (lower) the spi clock frequency, with no apparent change except that it increases data corruption, when there is any.

In my system there are no other SPI devices that can affect communication or other tasks in the firmware.

The code I use for the tests is the following, opening several instances of it.

```
#include <SPI.h>
#include <Ethernet.h>
#include <time.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(10, 42, 0, 177);
IPAddress myDns(10, 42, 0, 4);
IPAddress gateway(10, 42, 0, 4);
IPAddress subnet(255, 255, 0, 0);

EthernetServer server(23);

void setup() {
    // W5500 reset
    pinMode(9, OUTPUT);
    digitalWrite(9, LOW);
    delay(10);
    digitalWrite(9, HIGH);
    Ethernet.init(53);  // My board
    //Ethernet.init(10);  // Most Arduino shields
    Ethernet.begin(mac, ip, myDns, gateway, subnet);
    Serial.begin(9600);
    while (!Serial);
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
        Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
        while (1);
    }
    else Serial.println("Ethernet shield was found. =)");
    if (Ethernet.linkStatus() == LinkOFF) Serial.println("Ethernet cable is not connected.");
    else if (Ethernet.linkStatus() == LinkON) Serial.println("Ethernet cable is connected.");
    else Serial.println("Link status unknown.");
    server.begin();
    Serial.print("Chat server address:");
    Serial.println(Ethernet.localIP());
}

void loop() {
    static boolean alreadyConnected = false;
    char elapsed_time[64];
    static unsigned long t = 0;
    static time_t counter = 0;
    tm counter_time;

    EthernetClient client = server.available();

    if (client) {
        if (!alreadyConnected) {
            client.flush();
            Serial.println("We have a new client");
            client.println("Hello, client!");
            alreadyConnected = true;
        }
        if (client.available() > 0) {
            char thisChar = client.read();
            server.write(thisChar);
            Serial.write(thisChar);
        }
    }
    if (millis() - t >= 1000) {
        t = millis();
        counter++;
        memcpy(&counter_time, localtime(&counter), sizeof (struct tm));
        strftime(elapsed_time, 64, "Running for %H hours, %M minutes, %S seconds of day %j.", &counter_time);
        server.println(elapsed_time);
    }
}
```

Can you think of what might be happening? I've searched a lot about it and I'm surprised that I haven't found solutions but neither have I mentioned this kind of errors on a chip that is widely used. I am absolutely desperate and lost with this. I have seen that there are other issues that mention things similar to those that happen to me, but not all of them, and no solution yet.

SapientHetero, 2019-03-10:
A couple of quick observations - if you use the same HEX file on multiple devices on the same network, then each will have the same MAC and IP address. Each instance on the same network must have a unique MAC and a unique IP address in order to communicate. I suspect you'll find bit 7 of the W5500's Interrupt Register (IR) is set to indicate an IP Conflict has been detected.

Second, if you're using the Wiznet W5500 shield, it has a SD card reader on it. Whether you use it or not, you should set its CS pin as an output and set that pin to HIGH in setup() to prevent SPI communication problems.

Try addressing these 2 issues and see if it doesn't help.

kwendenarmo, 2019-03-11:
Thank you for your answer, both issues I have already taken in consideration:

- There is only one device per network, so there can not be collisions of MAC and IP.
- With respect to the two shields with socket for SD card:
    + For the shield of wiznet, there are no buffers for data signals, and they end up in an open terminal, so there is nothing that interferes with the SPI bus.
    + For the shield of Arduino, the MISO signal has no buffer, so it can not interfere. And as in the previous case, they end up in open terminals.
    + In addition, if this were the case, these boards with SD socket should be free of faults, which is not the case.

SapientHetero, 2019-03-29:
I know from your posts on the Wiznet W5500 forum that you had issues with overheated solder joints, but wondered if you still plan to experiment with my new version of the library?

kwendenarmo, 2019-03-30:
> I know from your posts on the Wiznet W5500 forum that you had issues with overheated solder joints, but wondered if you still plan to experiment with my new version of the library?

Yes, I plan to try it out when I finish a hardware design from another part of the same project.

MadDogMayCry0, 2022-02-21:
@kwendenarmo hi! Did you fix it? It's seems that my w5500 has same issue :(

kwendenarmo, 2022-02-21:
@MadDogMayCry0 This project was quite a while ago and I don't quite remember all the details, but I do know that a lot of the problems were solved when the chip was soldered at the factory, and not soldered manually. Apparently it is VERY sensitive to this, something that had not happened to me with the previous W5100.

What I know for sure is that all of this was fixed in the final version, not the prototype, and that the project has been running uninterruptedly for several years without incident.

shahkaush, 2024-05-28:
I am facing most of the issues mentioned by you.
Did you have to do any modifications to the Ethernet.h or W5100.h libraries? Or only the manual soldering was the culprit?

RileyStarlight, 2024-05-29:
As far as I remember, all the problems we had were solved when we manufactured with PCBA. I would say we didn't make any library modifications.

SapientHetero, 2024-05-30:
If you want to try my improved version of the W5500 library, you can find it at [https://github.com/SapientHetero/Ethernet](url)