arduino-libraries/Ethernet #94: Updated version of library

SapientHetero, 2019-03-22:
Just wanted to Ethernet 2.0 users know that an unofficial updated version of the library is available at https://github.com/SapientHetero/Ethernet that fixes many of the issues documented here and adds the ability to manage sockets in a more deterministic way. It builds on the great foundation provided by Paul Stoffregen and the team of contributors to this version, and I hope many find it useful.

Make sure to read README.rtf, which explains the changes made to each file in the library as well as how to use manageSockets().

denilly, 2019-10-09:
Good morning @SapientHetero, I'm having intermittent connectivity issues on SNMP and HTTP requests. It works for a while, but randomly lose connectivity, ping (ICMP) always responds. I have included server.begin () in the loop in which instability has improved, but connectivity drops remain, but the difference is that connectivity reestablishes itself. Because of this, I tried to use the Ethernet library you modified, but always has an error when compiling. Could help? Thankful!

denilly, 2019-10-09:
Follow errors while compiling ...

"C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:145:13: error: 'class W5100Class' has no member named 'waitForCmd'

  if (!W5100.waitForCmd(s, expected)) {

             ^~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp: In static member function 'static void EthernetClass::socketClose(uint8_t)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:267:13: error: 'class W5100Class' has no member named 'waitForCmd'

  if (!W5100.waitForCmd(s, SnSR::CLOSE_WAIT, SnSR::CLOSED, SnSR::FIN_WAIT)) {

             ^~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp: In static member function 'static uint8_t EthernetClass::socketListen(uint8_t)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:288:13: error: 'class W5100Class' has no member named 'waitForCmd'

  if (!W5100.waitForCmd(s, SnSR::LISTEN)) {

             ^~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp: In static member function 'static void EthernetClass::socketConnect(uint8_t, uint8_t*, uint16_t)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:308:13: error: 'class W5100Class' has no member named 'waitForCmd'

  if (!W5100.waitForCmd(s, SnSR::ESTABLISHED)) {

             ^~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp: In static member function 'static void EthernetClass::socketDisconnect(uint8_t)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:327:13: error: 'class W5100Class' has no member named 'waitForCmd'

  if (!W5100.waitForCmd(s, SnSR::TIME_WAIT, SnSR::CLOSED, SnSR::FIN_WAIT)) {   // wait for command to complete, as indicated by Socket n Status Register

             ^~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp: In static member function 'static uint32_t EthernetClass::getWaitTime()':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\socket.cpp:755:21: error: 'class W5100Class' has no member named 'getAvgWait'

  uint32_t t = W5100.getAvgWait();"

Rotzbua, 2019-10-09:
@denilly Better place for the issue, is directly on the repo: https://github.com/SapientHetero/Ethernet/issues

denilly, 2019-10-09:
I will comment there as requested. Thankful!

codeaholics, 2020-10-12:
@SapientHetero Thank you for this. This is the only version (out of literally dozens I've tried over the last few days) that can get my SeeedStudio W5500 working with a Mega 2560.

per1234, 2022-02-22:
Thanks for your work on the updated library and for sharing it here @SapientHetero.

closed 2022-02-22 by per1234