arduino-libraries/Ethernet #97: compilation error with ESP32 DIV Kit

pran111, 2019-04-24:
I Installed Ethernet 2.0.0 lib and try to compile example "webRepeatingClient" but it gaves following errors :

In file included from C:\Users\Akanksha\Documents\Arduino\libraries\Ethernet-2.0.0\examples\WebClientRepeating\WebClientRepeating.ino:26:0:

E:\Pranit\ArduinoInstalled\Arduino\libraries\Ethernet\src/Ethernet.h:258:2: error: invalid abstract return type 'EthernetClient'

  EthernetClient available();

  ^

E:\Pranit\ArduinoInstalled\Arduino\libraries\Ethernet\src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':

 class EthernetClient : public Client {

       ^

In file included from C:\Users\Akanksha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:157:0,

                 from C:\Users\Akanksha\AppData\Local\Temp\arduino_build_517046\sketch\WebClientRepeating.ino.cpp:1:

C:\Users\Akanksha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)

     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;

                 ^

C:\Users\Akanksha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)

     virtual int connect(const char *host, uint16_t port, int timeout) =0;

                 ^

In file included from C:\Users\Akanksha\Documents\Arduino\libraries\Ethernet-2.0.0\examples\WebClientRepeating\WebClientRepeating.ino:26:0:

E:\Pranit\ArduinoInstalled\Arduino\libraries\Ethernet\src/Ethernet.h:259:2: error: invalid abstract return type 'EthernetClient'

  EthernetClient accept();

  ^

WebClientRepeating:38:16: error: cannot declare variable 'client' to be of abstract type 'EthernetClient'

 EthernetClient client;

                ^

Can any one Face this issue.?

[image]

jandrassy, 2019-04-24:
in the ESP32 core changed they changed the Arduino API class Client. error is there
https://github.com/espressif/arduino-esp32/commit/9a9ff622166149da30dd13a81485184617c56387#commitcomment-33292990

thujer, 2019-05-14:
I use Platformio with VSCode and Arduino libraries. I have same problem with compilation Ethernet library for ESP32 platform. It lasts aproximately last week. Before it was ok and I no change libraries version. I tried to use an older or newer version of Arduino (nightbuild, 1.8.9, 1.8.0). Install Arduino with Platformio on clean OS (XUbuntu 16.04 and 18.04) under VirtualBox, but problem is still same. When I tried to downgrade C/C++ library in Platformio, after restart it install back last version (after reload I mean, that it use still last plugin version.

Have anybody similar experience?

How complicated can be repair this problem in Ethernet library?

[image]

jandrassy, 2019-05-14:
it is a problem of the ESP32 package. they still didn't fix it. create an issue there

thujer, 2019-05-14:
Thanks Jandrassy, but problem is still in any examples which use Ethernet library (not from ESP32 package), for example WebClient

[image]

jandrassy, 2019-05-14:
the ESP32 core has modified Arduino base class Client. it breaks compatibility by adding new pure virtual methods not implemented by the rest of the arduino world
https://github.com/espressif/arduino-esp32/issues/2755

thujer, 2019-05-15:
Jandrassy, I seen that You mentioned this problem in Espressif forum. Thanks

per1234, 2019-09-16:
Fixed by https://github.com/espressif/arduino-esp32/pull/3191

closed 2019-09-16 by per1234