arduino-libraries/Ethernet #99: SAMD51 (M4) based boards fail to communicate

BrunDog, 2019-05-23:
Hi,

Testing new SAMD51 (M4) based boards from Adafruit (Metro M4 and Grand Central M4, both based on the SAMD51)... using same simple chat server example.

Does not communicate with W5100 or W5500 based shields (Ethernet.hardwareStatus() returns EthernetNoHardware), Ethernet.localIP() returns no IP despite a static assignment, etc.

- Does communicate with the SD card on board both shields, so I am confident it is not a wiring issue or bus (plus nothing to wire as its a straight shield plug-in). Additionally, exact code works correctly with M4 boards using Ethernet2 library.

- Identical code works correctly on MEGA2560, Due, Uno with both W5100 and W5500 shields. 

This leads me to believe there is an incompatibility with SAMD51 & W5x00.

BrunDog, 2019-05-24:
Figured out the issue... w5100.h and w5100.cpp do not reference the SAMD51. I added definitions for this micro, which solved the problem.

That said, the library does need to be updated - I just don't know how to do that. Will let the admins do it.

dondipietro, 2020-04-27:
Hello Can you explain with modification you made ?
Do you added this ?

#elif defined(__SAMD51__)
	static volatile uint32_t *ss_pin_reg;
	static uint32_t ss_pin_mask;
	inline static void initSS() {
		ss_pin_reg = portModeRegister(digitalPinToPort(ss_pin));
		ss_pin_mask = digitalPinToBitMask(ss_pin);
		pinMode(ss_pin, OUTPUT);
	}
	inline static void setSS() {
		*(ss_pin_reg+5) = ss_pin_mask;
	}
	inline static void resetSS() {
		*(ss_pin_reg+6) = ss_pin_mask;
	}

bluespider42, 2020-07-01:
Does this library work with the M4 boards? what changes need to be made?