arduino-libraries/Ethernet #103: Problem with long prints (over 2048 bytes)

nightswimmer, 2019-06-05:


nightswimmer, 2019-06-05:
Hello. I'm having a problem when printing long blocks of text with this library when using a Teensyduino 3.2. I noticed the problem because I'm developing a webserver for Arduino which has the option to store full web pages in variables, and I have problems when I make prints over 2k bytes on the teensy. They work fine on the Arduino Mega.
The problem can be reproduced using the AdvancedChatServer example and making a newClient.print("some text over 2048 bytes....."). An arduino Mega will print the whole text, but the Teensy will only print the first 2048 bytes.

bleckers, 2020-06-14:
Probably a bit late to the party, but this is due to the buffer size setting in Ethernet.h:
```

// Configure the maximum number of sockets to support.  W5100 chips can have
// up to 4 sockets.  W5200 & W5500 can have up to 8 sockets.  Several bytes
// of RAM are used for each socket.  Reducing the maximum can save RAM, but
// you are limited to fewer simultaneous connections.
#if defined(RAMEND) && defined(RAMSTART) && ((RAMEND - RAMSTART) <= 2048)
#define MAX_SOCK_NUM 4
#else
#define MAX_SOCK_NUM 8
#endif

// By default, each socket uses 2K buffers inside the Wiznet chip.  If
// MAX_SOCK_NUM is set to fewer than the chip's maximum, uncommenting
// this will use larger buffers within the Wiznet chip.  Large buffers
// can really help with UDP protocols like Artnet.  In theory larger
// buffers should allow faster TCP over high-latency links, but this
// does not always seem to work in practice (maybe Wiznet bugs?)
//#define ETHERNET_LARGE_BUFFERS
```