arduino-libraries/Ethernet #106: Incorrect TCP packets when using W5500 with ESP32

maniekq, 2019-07-15:
While using ESP32 with W5500 it sometimes sends malformed packet. Time periods between malformed packets differs significantly - from few seconds to couple of hours. 
I've found this problem when using PubSubClient for MQTT where it caused client to be disconnected from MQTT broker.

To isolate the problem I've prepared very simple TCP server and I've been sending TCP messages to it with ESP32+W5500 in a single connection, continuously. It usually starts failing after few hours, then it breaks from time to time (sometimes within seconds, sometimes works for hours). I do not have this problem with ESP32 WiFi connection.

I've tried with another ESP32 and another W5500 with the same effect.

Please find Wireshark file with section of captured packages attached (packet number 15 is the problematic one).
[raw_TCP_incorrect_incoming_packet.zip](https://github.com/arduino-libraries/Ethernet/files/3393730/raw_TCP_incorrect_incoming_packet.zip)

### Hardware used:
DOIT Esp32 DevKit v1
Wiznet W5500

### Additional information
TCP server is set up in local network.
Simple implementation used to reproduce problem: https://github.com/maniekq/ESP32_W5500_TCP/blob/master/src/main.cpp#L73
Simple TCP server implementation: https://github.com/maniekq/servertcp-esp32

maniekq, 2019-07-25:
Is there any additional information I could add to make it more clear?
It would help me if someone could check if this is reproducible.

Rotzbua, 2019-07-26:
>  It would help me if someone could check if this is reproducible.

I can not help you, but if it should be "reproducible" you have to describe your wiring and build environment (versions).

maniekq, 2019-07-27:
Thanks @Rotzbua. I've added now missing information.

jcmertz, 2019-11-07:
I have been having this same issue repeatedly I believe, but hadn't gotten to this level of diagnosis! Have you had any luck finding a solution?

maniekq, 2019-11-07:
Unfortunately, I haven't found any solution yet. I've spent a lot of time trying to figure it out, but without good results. It happens quite rarely and 
occurs with random frequency which makes it really hard to track down. I am using WiFi connection for now, but I would like go back to this at some point.

SjefFransen1, 2020-03-27:
Unfortunately, I run in the same problem, and not yet found a solution for it? Somebody else with this problem?

mathieu56000, 2020-04-10:
Hi,
I think that I had the same problem and I kind of find a solution. Here is the same message that I posted in another post of yours maniekq. here is the link to my answer : 
https://github.com/aetas/RoomHub/issues/9#issuecomment-611933112

jandrassy, 2020-04-10:
> Hi,
> I think that I had the same problem and I kind of find a solution. here is the link to my answer :
> [aetas/RoomHub#9 (comment)](https://github.com/aetas/RoomHub/issues/9#issuecomment-611933112)

so powering. if the module can be powered with 5 V from USB, it works. if it is powered with the ESP32 from 3.3 V regulator, it has brown-outs

mathieu56000, 2020-04-10:
> so powering. if the module can be powered with 5 V from USB, it works. if it is powered with the ESP32 from 3.3 V regulator, it has brown-outs

Actually, I am powering both of the modules with a 5V regulator witch give enough power. So the ESP32 don't directly power the module. So I don't think this is a powering problem.

jandrassy, 2020-04-10:
> > > Hi,
> > > I think that I had the same problem and I kind of find a solution. here is the link to my answer :
> > > [aetas/RoomHub#9 (comment)](https://github.com/aetas/RoomHub/issues/9#issuecomment-611933112)
> > 
> > 
> > so powering. if the module can be powered with 5 V from USB, it works. if it is powered with the ESP32 from 3.3 V regulator, it has brown-outs
> 
> Actually, I am powering both of the modules with a 5V regulator which give enough power. So the ESP32 don't directly power the module. So I don't think this is a powering problem.

you power the small wiz850io compatible module with 5 V?

mathieu56000, 2020-04-11:
> > > > Hi,
> > > > I think that I had the same problem and I kind of find a solution. here is the link to my answer :
> > > > [aetas/RoomHub#9 (comment)](https://github.com/aetas/RoomHub/issues/9#issuecomment-611933112)
> > > 
> > > 
> > > so powering. if the module can be powered with 5 V from USB, it works. if it is powered with the ESP32 from 3.3 V regulator, it has brown-outs
> > 
> > 
> > Actually, I am powering both of the modules with a 5V regulator which give enough power. So the ESP32 don't directly power the module. So I don't think this is a powering problem.
> 
> you power the small wiz850io pinout compatible module with 5 V?!

I have no words ........ ^^" Yes, I was powering the small w5500 with 5V and he seemed to survived because now, since I am powering it with 3.3V, I don't have any deconnexion .. ^^" 
So forget about all I said, I haven't found any solution but in my case, the mqtt connexion seems to be very stable so maybe, some libraries were updated since the beginning of that topic.

ale-trevizoli, 2020-04-21:
Hi all.
Someone have tested it before using another 3.3v supplier and connecting gnd of esp32 and other supplier together? 
Anyone can test it?

mathieu56000, 2020-04-22:
Hi @ale-trevizoli, If you are using a external supplier, you don't has other choice but to connect the GNDs together (All GNDs of an installation should be connected together). This is what I have done and it works perfectly.