arduino-libraries/Ethernet #108: library looses incoming packets

geetee24, 2019-08-11:
brand new arduino branded ethernet 2 with using your latest code.

my app functions as web server.

when client sends large file, web server gets client.available  to false at times due to appears your library cant keep up with incoming http data.

is there a way to not lose packets?

SapientHetero, 2019-08-12:
It is not unusual for client.available() to return false and this does not indicate the loss of packets. 

Your server should wait for client.available() != 0 and implement a timeout so it doesn't hang if the client has stopped sending. For example:

`
int start = millis();
while (!client.available()) {
  if (millis() - start > timeout_ms)
    break;
}
...
`
TCP/IP includes a metering mechanism (window size) to prevent buffer overruns; this is handled by the Wiznet Ethernet chips and not the library.

Hope that helps.

geetee24, 2019-08-12:
it does exactly that is is losing chars

SapientHetero, 2019-08-12:
I don't doubt that your server is losing data; just noting that client.available() == 0 isn't an indication of lost packets.

It would be easier to help if you posted what hardware you're running on and the relevant portions of your code.