arduino-libraries/Ethernet #110: v2.0.0 of lib functoning as web server - browser client gets cant connect

geetee24, 2019-08-22:
i have simple server app.   
see: 
// https://wiznetmuseum.com/portfolio-items/lets-build-a-functional-arduino-webserver-with-the-ethernetsd-card-shield/?portfolioID=11350
// https://www.hackshed.co.uk/lets-build-a-functional-arduino-webserver-with-the-ethernet-sd-card-shield/

that app fails when a browser requests index.htlm that inside it contains.

<!DOCTYPE html>
<html>
<head>
	<base href="/" >
	<script src="scripts/js/ajs-app-98034266.js"></script>
	<script src="assets/config/configData-98034266.js"></script>
	<script src="scripts/js/vendor-98034266.js"></script>
</head>
<body>argh</body>
</html>

in chrome debugger you see it trying to open the 3 files at once and the bug is the 3rd file request immediately fails with err_connection_refused yet the 1st two download fine.

i think your library is not handling simultaneous connections?

Any help please.

jandrassy, 2019-08-23:
how are the request handled in sketch?

geetee24, 2019-08-23:
the library link i gave you is a super simple example that if you run it, you will see the exact error in the core library.   use any browser and see it.  I also see other google posts of the same problem with the library.  Thanks.  Note that I wrote my own much more feature rich web server and when I ported it to use your library, the same core issue surfaced.

geetee24, 2019-08-23:
see similar post here.
https://arduino.stackexchange.com/questions/34743/ethernet-shield-multiple-simultaneous-connections-cause-tcp-reset

jandrassy, 2019-08-23:
sorry but I don't see any code on that page in FF, Chrome or IA. There is "The following code should be altered to suit your setup:", but no code.

That Arduino SE Question is old. Which version of the library do you use?

jandrassy, 2019-08-23:
do you know this library https://github.com/UIPEthernet/UIPEthernet ?

jandrassy, 2019-08-23:
OK I found the code on the Wiznet page. It is inefficient, so slow and the browser times-out

jandrassy, 2019-08-23:
:-) I don't maintain this library (but I am a maintainer of the UIPEthernet library)

geetee24, 2019-08-23:
i will try that for the uip chip.  still need this library fixed :(   does your library support the zip app that I attached?

geetee24, 2019-08-23:
i tried your uiipard lib and it works great.  Thanks.   now i need THIS etrhernet lib fixed else the arduino ethner 2 card is worthless.

jandrassy, 2019-08-23:
write a better Web server. use buffering and caching headers. here is the [Web server in my project](https://github.com/jandrassy/Regulator/blob/master/Regulator/WebServer.ino). It uses my StreamLib for buffering.

geetee24, 2019-08-23:
what hardware and what lib does it use?

geetee24, 2019-08-23:
where is streamlib documented?

jandrassy, 2019-08-23:
> what hardware and what lib does it use?

almost any. Ethernet, UIPEthernet, esp8266, WiFiLink were used months. WiFiNina, WiFiEspAT were tested

> where is streamlib documented?

https://github.com/jandrassy/StreamLib/blob/master/README.md

geetee24, 2019-08-23:
i reviewed your server.   where is NetServer  defined? Code?

jandrassy, 2019-08-23:
> i reviewed your server. where is NetServer defined? Code?

the entire folder is one IDE project. the main file is Regulator.ino

geetee24, 2019-08-23:
so where are you saying the issue I am having with ethernet lib is?   in my web server?  where in your regulator code do you address this issue and solve for it?

jandrassy, 2019-08-23:
> so where are you saying the issue I am having with ethernet lib is? in my web server? where in your regulator code do you address this issue and solve for it?

in WebServer.ino. for static file serving

1) buffering of the outgoing bytes with BufferedPrint, which wraps `client`
2) setting `Content-length` of response [here](https://github.com/jandrassy/Regulator/blob/682e150266d0673150aabb8599da64c544c76278/Regulator/WebServer.ino#L118)
3) setting caching header 'Expires' for browser [here](https://github.com/jandrassy/Regulator/blob/682e150266d0673150aabb8599da64c544c76278/Regulator/WebServer.ino#L128)

geetee24, 2019-08-23:
but how does that relate to the browser requesting multiple connections and gettting tcp resets from the arduino

jandrassy, 2019-08-23:
I wrote it earlier. The browser has a timeout until it expects a response. If you not handle the requests fast enough, some of the requests timeout. With cached files the requests for static files are not repeated next time.

you get tcp resets at first set of requests after reset of Arduino?

geetee24, 2019-08-23:
Yes

geetee24, 2019-08-23:
to see the issue in the ethernet library, do not enable browser caching - for that only masks the core issue - the bug in the library.

SapientHetero, 2019-08-31:
This library has no more than one socket listening at a time. When a connection is made, there will be NO sockets listening while the library commands the W5500 to begin listening on a new socket. Current browsers establish up to 6 simultaneous connections to a server, and if this happens faster than your code can respond, some connection attempts will time out.

I modified this library to manage sockets in a manner more useful for servers and uploaded the new version to https://github.com/SapientHetero/Ethernet. You can tell it how many sockets you want listening at any time. It tracks the "last time used" for each socket, and closes the "oldest" sockets when needed to ensure the specified number of listeners are available. 

I also fixed a number of bugs I found in this library that improve reliability.