arduino-libraries/Ethernet #111: Ethernet cable is not connected

bkelly13, 2019-08-26:
Arduino DUE, Arduino Ethernet Shield 2, IDE 1.8.8, windows 10
Application:  Web Server from David A Mellis and Tom Igoe,, The text of the code was copied from here
https://www.arduino.cc/en/Tutorial/WebServer

This appears to be a problem with function Ethernet.linkStatus() returning an incorrect status.  It declares that the Ethernet cable is not connected when it is.

I modified this app by putting code at the top of the loop to blink the LED_BUILTIN.  It does blink and the rate can be easily changed.  A solid indicator that the app is properly downloaded.

After a reset the serial port displays:  Ethernet cable is not connected.  

Setup:

My desktop is hard cabled to a D-Link switch which then connects to my cable modem. Again, a cable and not wireless. 
I started with the computer on port 1 of the D-Link and the Arduino on port 4.  
When the Ethernet cable is connected to the Arduino, check the lights at the Ethernet connection.
Looking at the connector from the perspective of the cable, the LED on the left is lit green solid.
The status light on the D-Link switch for that port is on.
The Arduino LED on the right side of the cable blinks amber irregularly.  
Disconnect the Ethernet cable and both go out.

Get another Ethernet cable, same results.

Move the desktop computer connection to port 4, I still have access to the internet.  Port 4 works, so does port 1.
Swap the cable that was between the Arduino and the D-Link with the cable between the D-Link and cable modem.  The desktop continues to have internet connectivity.  All three of the Ethernet cables are good.

Replace the Arduino DUE and Ethernet shield with another sitting on my desk just for this reason.
Same Results.
Plug in a 12 Volt power supply so that the Arduino is now powered from the USB.
Same Results.

My project needs two Arduinos with Ethernet in separate locations, plus one more set for development.  Open the box and fire up the third set of Arduino and Ethernet shield.
Same results.

A ping from my computer does not solicit a reply.

From my perspective, the function Ethernet.linkStatus() is beyond my access and control.  Unless I really dig deep into this thing. 

What might I try now?

Might it help to re-install the IDE.  Maybe I left something out.

bkelly13, 2019-08-26:
I was not able to figure out how to move my previous post to here, nor could I find this without the link provided in a reply to my first post.  Here is the post again.
I did delete all the Arduino code from Windows 10 and used the on line install stuff to install it.  It took me to the Microsoft Store.  (Why there? this is not a Microsoft product).
Regardless, after the install the symptoms are identical.  The same link blinks at the rate the code specifies, Chrome cannot connect. Open a cmd window and ping, no response.
Connect serial monitor port and it displays:
well crum.  That window will not allow me to ctl-c and copy the code.  So I type it in.
[quote]
<time>  Ethernet WebServer Exaple
<time> Ethernet cable is not connected
<time> server is at 192.168.1.177
<time> Exiting setup
[/quote]

I don't know what else I might try.

per1234, 2019-08-30:
[Quote from @bkelly13](https://github.com/arduino/Arduino/issues/9157#issuecomment-526413185):
>Problem found. The Arduino IP was 192.168.1.177 while the router was set to 192.168.0.1. Changed the Arduino to 192.168.0.177 at it began responding to the browser.
That was easy. Once I knew what to change.
Thank you for your time.

bkelly13, 2019-08-31:
Still a problem
My original problem was, probably and still is, the network mask did not allow my computer to go from 192.168.0.177 to 192.168.1.177.  Changing the IP address of the board corrected that problem.
However, when the app is started this series of messages is displayed
[quote]
15:12:38.453 -> Ethernet WebServer Example
15:12:38.996 -> Ethernet cable is not connected.
15:12:39.030 -> server is at 192.168.0.174
15:12:39.064 -> Exiting setup
[/quote]
But when Chrome is started and provided that address, the serial port lights up with 
[quote]
15:12:53.276 -> new client
15:12:53.276 -> GET / HTTP/1.1
15:12:53.309 -> Host: 192.168.0.174
15:12:53.342 -> Connection: keep-alive
… and more
[/quote]

And Chrome connects and displays the results expected.

The error message that the Ethernet cable is not connected is incorrect.  There is not client but the cable is connected.

bkelly13, 2019-09-22:
Update.  When everything is functional but Chrome (or probably any browser) has not accessed the Arduino, the serial monitor displayed the the following:
14:45:52.840 -> Ethernet WebServer Example
14:45:53.370 -> Ethernet cable is not connected.
14:45:53.410 -> server is at 192.168.0.173
14:45:53.450 -> Exiting setup

When the correct IP address was entered into Chrome, the connection was made and the serial port began displaying the expected messages.

The Ethernet handler/driver does not differentiate between the states of:
No Ethernet Cable Connected
and that of:
No Client connected

The distinction is significant.

SapientHetero, 2019-09-24:
According to Wiznet, it does. They say that the LNK bit in the PHYCFGR register indicates that a cable connected to a network is detected.

Now, if you plug in a cable that isn't connected to anything on the other end, the example sketch is going to tell you that no cable plugged in. Feel free to change it to "not physically connected to network" or whatever is more meaningful to you.

Note that this has nothing whatsoever to do with whether a client is connected.

paolovr70, 2020-05-11:
Hi, I have the same ISSUE. With ARDUINO DUE, Ethernet Shield w5500. The Ethernet webserver example works BUT at setup I receive the message "the cable is not connected" (but again the server works). Any idea of what is the problem? tjanks. p.

yangxu1231, 2023-11-10:
I couldn't even change the ip no matter what static IP I set. It always prints "server is at 192.80.63.98", this is not even an illegal IP address. My god, could anyone figure this out!