arduino-libraries/Ethernet #112: Library causes memory corruption

gschaden, 2019-09-09:
When using this library I got strange behavior  in combination with my timer interrupt code.

Calling EthernetUDP::parsePacket in loop modified my IO registers somtimes. I could not find the root cause, it seems to be time critical. Calling this function with delay seems to make the issue to occur less often. The problem is not related to reading data from the network just checking for the availability of data shows the problem.

Using the library from https://github.com/Wiznet/WIZ_Ethernet_Library solved the problem.

SapientHetero, 2019-09-11:
When you say "modified my IO registers sometimes" do you mean W5500 registers or processor registers? Which register(s) were affected?

PaulStoffregen, 2019-09-11:
For that matter, which processor or board was used?

@gschaden - can you give us more details?  How about a complete program which calls parsePacket() and together with step-by-step instructions for sending the UDP packets to the board+shield, demonstrates this problem?