arduino-libraries/Ethernet #116: Configure to HSPI on ESP32

jhnwmr, 2019-10-14:
Hello,
I want to use the W5100 Ethernet module with the ESP32 on the second SPI bus.
I also use the other bus. What change do I have to make in the library to use the second bus?

per1234, 2019-10-14:
This issue tracker is only to be used to report bugs or feature requests. It is not a place to request support. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

PaulStoffregen, 2019-10-15:
@per1234 - this kind of is a feature request.  It could be re-imagined about any board with more than one SPI port.  I would recommend reopening, but rename to something like "allow use of other SPI ports, on boards with multiple SPI".

Ralf9, 2020-01-20:
I can use this too.
Is this a usable option?
https://github.com/stevstrong/Ethernet_STM32

-  The library needs to know the SPI port on which you connected the W5500 chip, and the corresponding chip select pin. Therefore you have to declare an SPI class object with appropriate SPI port number (1..2 for F1, 1..3 for F4)
`SPIClass mSpi(1); // you can use 1..2 for F1, 1..3 for F4)`
- Then you need to call the following function in setup() before Ethernet.begin():
`Ethernet.init(mSpi, PA4); // SPI class object, chip select pin on your choice`

DprasK, 2023-07-05:
if i use mcu board (official not maple core)?