arduino-libraries/Ethernet #120: corrupted web server web pages on esp32 with W5500 module

Jay-esp, 2019-11-28:
does anybody use this combination with success?
I have constant corrupted web pages (some characters are wrong), even with the basic web server example
I do not have this issue with the ethernet2 library but that has other issues and is no longer maintained, my project is up and running for 6 months now without too many web issues but I would like to go back to the "official" ethernet library.

ale-trevizoli, 2020-01-13:
I have same problem too. Don't know how to resolve.

maniekq, 2020-04-16:
Have you managed to do any progress with this problem?
It sounds like what I have with MQTT and TCP connection in general: https://github.com/arduino-libraries/Ethernet/issues/106 .

ale-trevizoli, 2020-04-21:
No Solution yet. I check library Ethernet and datasheet of W5500 and it's
have some registers differences. I think that is where is the problem. I
tried to talk with  PaulStoffregen <https://github.com/PaulStoffregen>
Etthernet
developer, but he can't investigate because he don't like ESP32 platform.

ale-trevizoli, 2020-04-21:
I'm using very OLD ethernet library (before 2.0 release), It's not has this
bug, but has others.

Jay-esp, 2020-04-22:
if it's not fixed, the problem is still there :) 
I'm still using the ethernet2 library for now, hardware is not changing so it keeps on working

Jay-esp, 2020-04-22:
I did compare the code ethernet vs ethernet2 some time ago and the base code looked completely different so it's not easy to just hunt for the difference and track down the issues
we all appreciate Paul's hard work, I think he has his hands full with the teensy modules for now

maniekq, 2020-05-13:
@Jay-esp Could you point me to the library you use?
I've found these two:
https://github.com/adafruit/Ethernet2 or https://github.com/PaulStoffregen/Ethernet

Jay-esp, 2020-05-17:
using the adafruit one ethernet2

Alexandr-zz2not1, 2020-10-18:
https://github.com/PaulStoffregen/Ethernet/issues/39#issuecomment-711109000

ale-trevizoli, 2020-10-19:
Alexandr-zz2not1 yes your patch works. I did that and problem disappear. Seems like problem is in SPI library of ESP32.

ale-trevizoli, 2020-10-19:
Thanks a lot for share solution!

Pongpun364, 2020-10-20:
> Thanks a lot for share solution!

Could you share your solution ? please..