arduino-libraries/Ethernet #122: Ethernet library compile warning

RichardSelmanXX, 2019-12-18:
Reposted from main IDE.  IDE 1.8.10.  Controllino MEGA, simple ethernet test sketch......

Get this on compile....

In file included from C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp:8:0:

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]

 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )

                      ~~~^~~

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dns.cpp:164:18: note: in expansion of macro 'htons'

  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);

                  ^~~~~

per1234, 2019-12-18:
From the original issue https://github.com/arduino/Arduino/issues/9539: 
*matthijskooijman commented 32 minutes ago*
>Seems like a warning, which is also harmless (code works as expected AFAICS).
>
>Still would be good to fix, though.

thekunalsaini, 2020-03-30:
@RichardSelmanXX  Is there any documentation you have follow for this

5kuby, 2020-06-16:
is it still unfixed?

ckohrt, 2021-01-07:
Any news here?

mgnthe54, 2021-01-25:
Hello, for some time I have also been getting this alert message on one of my PCs ....
Anyone have any idea where the warning came from?
Thank you

ckohrt, 2021-01-25:
> Hello, for some time I have also been getting this alert message on one of my PCs ....
> Anyone have any idea where the warning came from?
> Thank you

I just ignore it.

mgnthe54, 2021-01-25:
I found this in another post, I did it and I no longer have an error message ... Thanks

https://forum.arduino.cc/index.php?topic=647823.0 :  post of ToneArt  of 04/04/2020

If you want to avoid the warning, just fix the w5100.h file (in <Arduino install> \ libraries \ Ethernet \ src \ Utility):

at the end of the file you have :
#define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )

replace it with this
#define htons(x) ( ((((x)&0xFF)<<8)&0xFF00) | (((x)>>8)&0xFF) )

However, I only had this error message on my desktop computer and not on my other portable computer!

closed 2021-07-22 by aentinger

evantill, 2022-07-04:
Still present using the latest version of the library (https://github.com/arduino-libraries/Ethernet/releases/tag/2.0.0)

PR #161 (2021-07-22): Fix compiler warning in `htons()`
Fixes #151
Fixes #122 

~~The macros are completely redundant. Using for example `htons()` on a `uint16_t` just wastes cycles if the compiler doesn't optimise it out. `htons()` converts a variable to 16 bits (which does nothing for a 16 bit variable). Same goes for `htonl()` for 32-bit variables.~~

~~Building these changes in a project of mine shows the memory footprint is reduced by **32 bytes**. This will also fix two issues related to overflow compiler warnings with `htons()`.~~

Edit: This PR will only fix the compiler warning in the linked issues.