arduino-libraries/Ethernet #123: UDP packet receive not working when changed to fixed IP

RichardSelmanXX, 2019-12-19:
Controllino MEGA.  IDE 1.8.10

Just setting up the Controllino to sync the RTC with a NTP source.  Downloaded a simple NTP example sketch and the system reported the correct time.  Integrated it with my main code and it stopped working.  Downloaded several other examples including this general [one](https://github.com/arduino-libraries/NTPClient) this controllino specific [one](https://github.com/ckuehnel/Controllino/blob/master/UdpNtpClient.ino) with same result.  Traced it down whether I have a fixed IP address or not.  Examples use DHCP and a ethernet.begin(mac) call.  If I change this to ethernet.begin(mac,IP) the general ethernet still works but I don't seem to get a return packet from the NTP server.  Could it be something to do with the DHCP call that must be occurring in the examples but not in my fixed IP case.   Not really competent to step through the ethernet.cpp to see what's different.  Any thoughts?

thekunalsaini, 2020-03-30:
Hi @RichardSelmanXX,
`
unsigned int localPort = 5000;

// An UDP instance to let us send and receive packets over UDP
UDP Udp;

void setup() {
    // Initialise UDP
    Udp.begin(5000);
    
    // Print your device IP Address via serial 
    Serial.begin(9600);

}

void loop() {
    // Check if data has been received
    Serial.println("At if loop");
    delay(2000);
    if (Udp.parsePacket() > 0) {
        Serial.println("Packet Recieved");
        delay(2000);
        // Read first char of data received
        char c = Udp.read();
        
        // Ignore other chars
        Udp.flush();
}
Sender:

UDP udp;

byte remoteIP[4] = { 172, 16, 0, 221};
int remotePort = 5000;

unsigned char TxMsg[5] = { 'H', 'E', 'L', 'L', 'O'};

void setup() {
    udp.begin(5000);
    Serial.begin(9600);
}

void loop() {

    Serial.println("Running");
    udp.beginPacket(IPAddress( 172, 16, 0, 221), 5000);
    udp.write(TxMsg, 5);
    udp.endPacket();
    delay(2000);

}
`
Try this on Uno. It will work

maxpromer, 2021-11-12:
I have same issues, How to fixed it ?