arduino-libraries/Ethernet #124: ESP32+W5500 corrupt data in tcp packets

ale-trevizoli, 2020-01-15:
I'm using this Paul library for long time and I love it, but now I migrate some codes to ESP32 with W5500 (berfore I using MEGA2560+W5100), but on W5500 some random times I'm getting corrupt tcp data packets, if we make WEBSERVER sample, and do a lot of refresh pages in Browser we get corrupt characters on page. 
I did tests using ESP32+W5100 (works with no problem)  and ESP32+W5500(with OLD Ethernet2 library) and it's not happen.
Could we fix it?
You can simulate only taking ESP32+W5500 and WEBSERVER example. 
I'm using pins GPIO 5(CS),18(CLK),19(MISO) and 23(MOSI).
W5500 Module:
https://ae01.alicdn.com/kf/H6e6dbc78b77c493fb11edd8e8fa3f4aec/USR-ES1-w5500-chip-novo-spi-para-lan-ethernet-conversor-tcp-ip-mod.jpg

Thanks for all!

Rotzbua, 2020-01-15:
seems to be a doublicate of #120

per1234, 2020-01-15:
Closing as duplicate of #120 (thanks @Rotzbua!).

@ale-trevizoli, I have copied your comment over to #120, since it may provide some additional useful information to that issue.

closed 2020-01-15 by per1234