arduino-libraries/Ethernet #127: HTTPS

BlaiMunoz, 2020-03-06:
Hi!

I am trying to implement your library to make an HTTPS POST using Arduino (I can do an HTTP POST, but I need to be HHTPS). I have found the library ArduinoBearSSL but I can make it work because there is this step:

 ArduinoBearSSL.onGetTime(getTime);

Where getTime is:
unsigned long getTime() {
  return WiFi.getTime();
}

Unfortunately, there is no getTime class in the Ethernet library.

How can this be solved? I would really appreciate your help.

jandrassy, 2020-03-06:
with the NTP library or see the UdpNtpClient example of the Ethernet library

per1234, 2020-03-06:
@BlaiMunoz, this issue tracker is only to be used to report bugs or feature requests. If you require further assistance, please post on the [Arduino Forum](http://forum.arduino.cc/index.php?topic=148850), I'm sure we'll be able to help you out over there.

closed 2020-03-06 by per1234