arduino-libraries/Ethernet #128: Definition of INADDR_NONE in header file bloats final executable size

MHotchin, 2020-03-07:
Currently, IPADDRESS.H both declares and defines INADDR_NONE.

If a sketch has more than one file that includes IPAddress.h, then multiple copies of INADDR_NONE end up in the final executable.  Essentially, each CPP file gets its own copy as a global variable, at 6 bytes a pop.

If the declaration in IPAddress.h is changed to:
extern const IPAddress INADDR_NONE;

and a new definition is provided in IPAddress.cpp:
const IPAddress INADDR_NONE(0, 0, 0, 0);

then only one copy end up in the executable.

My sketch went from 2002 bytes 'Minimum Memory Usage' to 1888 bytes, a savings of 78 bytes in my case.

Rotzbua, 2020-03-07:
The `ethernet` library does not contain a `IPAddress.h/.ccp` .

It is part of the main arduino framework. I reported this years ago. It is fixed in the default arduino framework.

My last update was that the uno is  not (yet?) migrated to the default arduino framework.

Best

per1234, 2020-03-07:
Thanks for the explanation @Rotzbua.

Closing as duplicate of https://github.com/arduino/Arduino/issues/1007, which was fixed by https://github.com/arduino/ArduinoCore-API/commit/3dbd7d7d338ff2d93c8b9dd8c833967fdb69b7c6 (but has not yet propagated out to all the core libraries).

closed 2020-03-07 by per1234