arduino-libraries/Ethernet #133: Server not rejecting client connections not destined for the server port

SheetLightning, 2020-05-09:
I have started a server with for example in the definitions:

`EthernetServer ethServer(8080);`

In void setup():

`ethServer.begin();
`
In the loop I am accepting data from a connected client:

```
      EthernetClient ethClient = ethServer.available();
      if (ethClient){
        if (ethClientConnected){
          while (ethClient.available()) {
            // Read the bytes incoming from the client:
            char chr = ethClient.read();
            // Echo the bytes to serial
            Serial.write(chr);
          }
        }
      }else{
        if (ethClientConnected){
          ethClient.stop();
          ethClientConnected = false;
        }
      }
  }
```
However, I found that it is also possible to connect with any other TCP port. The client will not accept or receive data on that port and will eventually time out, but the client (PuTTY) is establishing a connection rather than being immediately rejected. Tried with both RAW and Telnet options.

Rotzbua, 2020-05-09:
Hint: Do not forget to call `maintain()` in loop.

jandrassy, 2020-05-09:
> Hint: Do not forget to call `maintain()` in loop.

why? it only handles the DHCP lease

jandrassy, 2020-05-09:
I can't replicate it. I get  "Connection refused".

```
duro@nuc ~ $ telnet 192.168.1.6 2323
Trying 192.168.1.6...
Connected to 192.168.1.6.
Escape character is '^]'.

16:27:31;11019;R100;308;-3;98;-18;279;787;372;636;236;387;
16:27:35;11019;R100;329;37;98;-16;286;759;346;625;236;366;
C
Connection closed by foreign host.
duro@nuc ~ $ telnet 192.168.1.6 23
Trying 192.168.1.6...
telnet: Unable to connect to remote host: Connection refused
```

jandrassy, 2020-05-09:
the test in my previous comment was with W5500.
with W5100 I could replicate the issue.

SheetLightning, 2020-05-09:
Thanks for your tests with the W5500 vs the W5100. The results of your test with the W5500 is how I would expect it to behave, but as you have now confirmed, the W5100 behaves differently. Remains to be determined how the W5200 behaves.

Rotzbua, 2020-05-10:
Does the problem also occur with old library version v1.1.2? Maybe something went wrong in v2.

SheetLightning, 2020-05-11:
Just tested that and unfortunately yes, it does.

SheetLightning, 2020-05-25:
I raised a case with the people over at WizNet and having investigated the problem, they seem to think this is related to the Arduino library:

> I think we or you need to check the Ethernet library in Arduino IDE.
> Because this issue didn’t occur in the io-library code we provide,it is for embedded user.

I do hope this can be fixed in the library?

https://forum.wiznet.io/t/topic/7470/3