arduino-libraries/Ethernet #146: error: 'class IPAddress' has no member named '_address'

eg321, 2020-09-01:
Hi,

There are compilation errors at ESP8266 with statically defined IP:

```
.pio\libdeps\d1_mini\Ethernet\src\Ethernet.cpp:87:24: error: 'class IPAddress' has no member named '_address'
  W5100.setIPAddress(ip._address.bytes);
                        ^
.pio\libdeps\d1_mini\Ethernet\src\Ethernet.cpp:88:29: error: 'class IPAddress' has no member named '_address'
  W5100.setGatewayIp(gateway._address.bytes);
                             ^
.pio\libdeps\d1_mini\Ethernet\src\Ethernet.cpp:89:29: error: 'class IPAddress' has no member named '_address'
  W5100.setSubnetMask(subnet._address.bytes);
```

It's quite strange, because "_address" field was removed almost 2 years ago from "IPAddress" class: https://github.com/esp8266/Arduino/commit/5c4db3acf4353870b6a5663ba06d712d67e6c179#diff-8443440f4d240d2a531dfedcec06baaa

So, dirty fix is quite easy:
```
	W5100.setIPAddress((uint8_t *)&ip.v4());
	W5100.setGatewayIp((uint8_t *)&gateway.v4());
	W5100.setSubnetMask((uint8_t *)&subnet.v4()); 
```

but maybe we should think about ipv6 here also...

jandrassy, 2020-09-01:
esp8266 Arduino boards package has old version of Ethernet library.

eg321, 2020-09-01:
Hm...
I'm using latest Platformio's ESP8266 2.6.2 platform.  It's based on Arduino core 2.7.4 as I can see.
Is it Platformio specific issue?

jandrassy, 2020-09-01:
> Hm...
> I'm using latest Platformio's ESP8266 2.6.2 platform. It's based on Arduino core 2.7.4 as I can see.
> Is it Platformio specific issue?

I don't know how platformio bends arduino packages and libraries.

eg321, 2020-09-22:
> > Hm...
> > I'm using latest Platformio's ESP8266 2.6.2 platform. It's based on Arduino core 2.7.4 as I can see.
> > Is it Platformio specific issue?
> 
> I don't know how platformio bends arduino packages and libraries.

It looks like Ethernet library was not released after https://github.com/arduino-libraries/Ethernet/commit/0d5019a412857e6b221dd45c2c93633c7b62b276#diff-1d2d2b6cbd22874ff55b7362b0babf39 fix.

It was commited at 20 Apr 2019, but Ethernet library latest release is v2.0 - 26 Jul 2018.
So, latest release is not compatible with ESP8266 now.

Can you include the fix above into release please?

eg321, 2020-09-22:
@PaulStoffregen it seems you have enough privileges. Can you assist with that please?

jandrassy, 2020-09-22:
> 
> 
> > > Hm...
> > > I'm using latest Platformio's ESP8266 2.6.2 platform. It's based on Arduino core 2.7.4 as I can see.
> > > Is it Platformio specific issue?
> > 
> > 
> > I don't know how platformio bends arduino packages and libraries.
> 
> It looks like Ethernet library was not released after [0d5019a#diff-1d2d2b6cbd22874ff55b7362b0babf39](https://github.com/arduino-libraries/Ethernet/commit/0d5019a412857e6b221dd45c2c93633c7b62b276#diff-1d2d2b6cbd22874ff55b7362b0babf39) fix.
> 
> It was commited at 20 Apr 2019, but Ethernet library latest release is v2.0 - 26 Jul 2018.
> So, latest release is not compatible with ESP8266 now.
> 
> Can you include the fix above into release please?

the esp8266 arduino has (will have) this library in package libraries as subproject

eg321, 2020-09-22:
> the esp8266 arduino has (will have) this library in package libraries as subproject

Can you provide some link / issue for reference please? ETA is very appreciated.

jandrassy, 2020-09-22:
> > the esp8266 arduino has (will have) this library in package libraries as subproject
> 
> Can you provide some link / issue for reference please? ETA is very appreciated.

it is already in the GitHub repository of the esp8266 arduino, but the next release from the master branch will be the 3.00 and it will not be ready soon.
btw the 3.00 will have 'native' support for W5500 (and enc28j60). it will be a network interface like the STA or SoftAP, so it will be used with WiFiClient, WiFiServer, WiFiUdp, ESP8266WebServer, OTA ...

until then you can download the zip of this repo and install it as library. and don't forget to delete the Ethernet library bundled with the esp8266 boards package