arduino-libraries/Ethernet #151: Overflow warning in htons()

habazut, 2020-09-27:
In file included from /usr/share/arduino-1.8.13/libraries/Ethernet/src/Dns.cpp:8:0:
/usr/share/arduino-1.8.13/libraries/Ethernet/src/Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':
/usr/share/arduino-1.8.13/libraries/Ethernet/src/utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]
 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )

+++ Set a pragma to locally turn of that warning if you know about the overflow and it's OK. +++

Your library shall compile without warnings.

Regards,
Harald.

jmnc2, 2021-01-07:
Is this lib not getting bug fixes any more?

habazut, 2021-01-31:
Suggestion: 
#define htons(x) ( ((( x )&0xFF)<<8) | ((( x )&0xFF00)>>8) )

and for #pragma see google.

Regards,
Harald.

GunZi200, 2021-04-19:
I am seeing this also: 

[image]

GunZi200, 2021-04-19:
The real issue is line 164 in DNS.cpp

`twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);`

change this to:

`twoByteBuffer = RECURSION_DESIRED_FLAG;`

Using `htons()` here makes little sense to me. Since `htons()` combines two bytes together while `RECURSION_DESIRED_FLAG` is part of a 16-bit mask.

PR #161 (2021-07-22): Fix compiler warning in `htons()`
Fixes #151
Fixes #122 

~~The macros are completely redundant. Using for example `htons()` on a `uint16_t` just wastes cycles if the compiler doesn't optimise it out. `htons()` converts a variable to 16 bits (which does nothing for a 16 bit variable). Same goes for `htonl()` for 32-bit variables.~~

~~Building these changes in a project of mine shows the memory footprint is reduced by **32 bytes**. This will also fix two issues related to overflow compiler warnings with `htons()`.~~

Edit: This PR will only fix the compiler warning in the linked issues.
closed 2021-07-22 by aentinger