arduino-libraries/Ethernet #169: ESP32 + W5500 (Ethernet Shield 2) is not recognised

barcesat, 2021-09-28:
Hello all,
I have tried many combinations but I just can't understand what's the problem.
I have an ESP32 Dev module that I have successfully connected to the old W5100 shield in this configuration:

|ESP32 | SHIELD|
| --- | --- |
|GPIO18 | SCK (ICSP connector)|
|GPIO19| MISO (ICSP connector)|
|GPIO23| MOSI (ICSP connector)|
|GPIO5| CS (pin10)|
|EN|RESET|
|5V|5V|
|GND|GND|

I have tried to use the same code by selecting the CS pin correctly with 'Ethernet.init(5);'.
However, the shield appears to not be recognized by the ESP32 and thus the connection is failing.
To test whether the shield is defective I have tried it successfully with Arduino Uno (CS pin=10 of course).
I also tested another exact shield with no success.

I thought that perhaps the reset is not connected properly so I've tried connecting it to a free GPIO (32) and run this WizReset function (that I've found somewhere) between Ethernet.init(5); & Ethernet.begin(mac). 

`
void WizReset(int reset_pin) {
    MySerial.print("Resetting Wiz W5500 Ethernet Board...  ");
    pinMode(reset_pin, OUTPUT);
    digitalWrite(reset_pin, HIGH);
    delay(250);
    digitalWrite(reset_pin, LOW);
    delay(50);
    digitalWrite(reset_pin, HIGH);
    delay(350);
    MySerial.println("Done.");
}
`
It also didn't work...
I also tried connecting the reset pin straight to 3.3V but that also failed.
I tried all kinds of forks of this library (including the infamous Ethernet2) without seccess.
Does anyone have any idea what am I doing wrong?
Thanks!

lathoub, 2021-10-02:
I have an ESP32 WROOM32 talking happily to a W5500, using the std Ethernet library with CS pin on 5 of the ESP32 and SCL on PIO22 (not on your drawing)

Only 'special' code I had to add was:
`Ethernet.init(5);`

So **no** extra resetting of the W5500

philm001, 2024-01-05:
@lathoub Can you share your connections? 

I have an ESP32 DevKit C and I also am unable to get the v2 of the Ethernet shield working. I am able to have it work with the v1 which uses the 5100 chip. I am pretty sure that the connections are the same. Will be testing a little more and posting my issue soon.

philm001, 2024-01-05:
So here is where the issue is:

```
uint8_t W5100Class::isW5500(void)
{
	chip = 55;
	Serial.println("w5100.cpp: detect W5500 chip");
	if (!softReset()) return 0;
	writeMR(0x08);
	int someValue = readMR();
	Serial.println(String(someValue));
	if (someValue != 0x00) return 0;
	writeMR(0x10);
	someValue = readMR();
	Serial.println(String(someValue));
	if (someValue != 0x10) return 0;
	writeMR(0x00);
	if (readMR() != 0x00) return 0;
	int ver = readVERSIONR_W5500();
	//Serial.print("version=");
	//Serial.println(ver);
	if (ver != 4) return 0;
	//Serial.println("chip is W5500");
	return 1;
}

In here, both of the read functions return a 0
```