arduino-libraries/Ethernet #186: Add support to change HOST_NAME constant

sezeryalcin, 2022-03-21:
There is absolutely no point of keeping a static hostname prefix as this library is free and open source. There are already made patches for allowing customization of DHCP hostname string so please allow this functionality.

>Dhcp.h
..
#define HOST_NAME "WIZNet"
I am seeing following in my DHCP table which could make much sense if I had  a chance to customize it.

WIZnetEFFEEC | DE:AD:BE:EF:FE:EC | 192.168.1.107 | 47:49:21
WIZnetEFFEEB | DE:AD:BE:EF:FE:EB | 192.168.1.108 | 47:50:28
WIZnetEFFEEA | DE:AD:BE:EF:FE:EA | 192.168.1.110 | 47:58:33
WIZnetEFFEE0

Russell108, 2022-11-28:
Hello Arduino team I have many Arduino on my network If I can set the hostname as in the wifinina library, it would be really easy to find them on my DHCP server
It's not a trivial thing but really useful for administration

Russell108, 2022-12-08:
OK For me, this is no longer such a big issue I realise that the hostname is unique
I understand that the default will be WIZnet, followed by the last three octets of the mac address.

thirstyice, 2024-01-23:
See my PR #248