arduino-libraries/Ethernet #193: Error when compiling "WebServer" example for ESP32 Dev Module

themowa, 2022-05-09:
I want to create a Webserver using my ESP32 with W5500 Board. My Setup works but when i try to create a Webserver i get this error:

WebServer:33:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'
 EthernetServer server(80);
                ^
In file included from C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\Ethernet\examples\WebServer\WebServer.ino:21:0:
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\Ethernet\src/Ethernet.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^
In file included from C:\Users\morit\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:152:0,
                 from sketch\WebServer.ino.cpp:1:
C:\Users\morit\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^
exit status 1
cannot declare variable 'server' to be of abstract type 'EthernetServer'

:(
Can I fix this somehow?
BTW It compiles fine for arduino boards

per1234, 2022-05-09:
Hi @themowa. Thanks for your report. I see we have a previous discussion of the problem here: https://github.com/arduino-libraries/Ethernet/issues/88

There you can find the change that needs to be made to ESP32's Arduino core to fix the bug: https://github.com/arduino-libraries/Ethernet/issues/88#issuecomment-455498941

I don't think there is anything we can do about the problem here, which is why I accepted the closure of https://github.com/arduino-libraries/Ethernet/issues/88. As you noted, this is specific to the 3rd party ESP32 boards platform, which Arduino does not maintain.

They had an issue to track the problem at https://github.com/espressif/arduino-esp32/issues/2704, but unfortunately it was closed by their automated system. That issue was recently marked as "duplicate" by @VojtechBartoska, so maybe there has been some progress I am not seeing after a quick scan of the repository? I'm not certain which issue it is considered a duplicate of. Perhaps it is https://github.com/espressif/arduino-esp32/issues/6058, which came years later and was also closed without a resolution?

If you end up with additional information to share, feel free to comment in https://github.com/arduino-libraries/Ethernet/issues/88. In that way we can consolidate all relevant discussion to one place.

closed 2022-05-09 by per1234

JAndrassy, 2022-05-09:
a quick fix would be to change in Ethernet.h the line
`class EthernetServer : public Server {`
to
`class EthernetServer : public Stream {`