arduino-libraries/Ethernet #194: Ethernet cable is not connected

stevenlianyi, 2022-05-10:
Initialization (begin()) adds soft reset to prevent the switch from not detecting the cable. 
Situation: Using WIZ820io RC5 module, ESP32 control
I found that if the switch and the module are powered on at the same time, there is a slight chance that the module cannot get the IP address, and if I observe the switch, I find that the corresponding port green light is off and cannot recognize the module.
Solution: Modify Ethernet.h/Ethernet.cpp to add software reset at the beginning. Problem solved.

admindotnu, 2022-05-20:
We experience same behavior with the w5500 chips.

admindotnu, 2022-05-20:
> Initialization (begin()) adds soft reset to prevent the switch from not detecting the cable. Situation: Using WIZ820io RC5 module, ESP32 control I found that if the switch and the module are powered [...]

We experience same behavior with the w5500 chips.

Could you share what u changed ?